ARMmbed/mbed-os #2306: K22F tests fail - ARM OoB RC#3

toyowata, 2016-07-29:
Version: https://github.com/mbedmicro/mbed/releases/tag/mbed-os-5.1.0-rc3
ARM toolchain
Target: K22F
Host OS: Windows 7

```

mbedgt: test suite report:
+----------+---------------+-------------------------------------------------------------------------------------------+---------+--------------------+-------------+
| target   | platform_name | test suite                                                                                | result  | elapsed_time (sec) | copy_method |
+----------+---------------+-------------------------------------------------------------------------------------------+---------+--------------------+-------------+
| K22F-ARM | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-basic_test                        | OK      | 9.58               | shell       |
| K22F-ARM | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-basic_test_default                | OK      | 9.51               | shell       |
| K22F-ARM | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-case_async_validate               | OK      | 11.18              | shell       |
| K22F-ARM | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-case_control_async                | OK      | 17.37              | shell       |
| K22F-ARM | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-case_control_repeat               | OK      | 11.56              | shell       |
| K22F-ARM | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-case_selection                    | OK      | 9.33               | shell       |
| K22F-ARM | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-case_setup_failure                | OK      | 9.91               | shell       |
| K22F-ARM | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-case_teardown_failure             | OK      | 10.03              | shell       |
| K22F-ARM | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-control_type                      | OK      | 10.19              | shell       |
| K22F-ARM | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-minimal_async_scheduler           | OK      | 10.19              | shell       |
| K22F-ARM | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-minimal_scheduler                 | OK      | 10.85              | shell       |
| K22F-ARM | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-test_assertion_failure_test_setup | OK      | 9.28               | shell       |
| K22F-ARM | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-test_setup_case_selection_failure | TIMEOUT | 67.17              | shell       |
| K22F-ARM | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-test_setup_failure                | TIMEOUT | 67.06              | shell       |
| K22F-ARM | K22F          | mbed-os-core-tests-integration-basic                                                      | OK      | 8.46               | shell       |
| K22F-ARM | K22F          | mbed-os-core-tests-integration-threaded_blinky                                            | OK      | 28.47              | shell       |
| K22F-ARM | K22F          | mbed-os-core-tests-mbed_drivers-c_strings                                                 | OK      | 11.48              | shell       |
| K22F-ARM | K22F          | mbed-os-core-tests-mbed_drivers-callback                                                  | TIMEOUT | 67.89              | shell       |
| K22F-ARM | K22F          | mbed-os-core-tests-mbed_drivers-dev_null                                                  | OK      | 10.63              | shell       |
| K22F-ARM | K22F          | mbed-os-core-tests-mbed_drivers-echo                                                      | OK      | 20.41              | shell       |
| K22F-ARM | K22F          | mbed-os-core-tests-mbed_drivers-generic_tests                                             | OK      | 9.83               | shell       |
| K22F-ARM | K22F          | mbed-os-core-tests-mbed_drivers-rtc                                                       | TIMEOUT | 67.09              | shell       |
| K22F-ARM | K22F          | mbed-os-core-tests-mbed_drivers-stl_features                                              | OK      | 10.58              | shell       |
| K22F-ARM | K22F          | mbed-os-core-tests-mbed_drivers-ticker                                                    | OK      | 40.49              | shell       |
| K22F-ARM | K22F          | mbed-os-core-tests-mbed_drivers-timeout                                                   | TIMEOUT | 67.05              | shell       |
| K22F-ARM | K22F          | mbed-os-core-tests-mbed_drivers-wait_us                                                   | OK      | 18.99              | shell       |
| K22F-ARM | K22F          | mbed-os-core-tests-mbedmicro-mbed-attributes                                              | OK      | 11.13              | shell       |
| K22F-ARM | K22F          | mbed-os-core-tests-mbedmicro-mbed-call_before_main                                        | OK      | 8.61               | shell       |
| K22F-ARM | K22F          | mbed-os-core-tests-mbedmicro-mbed-cpp                                                     | OK      | 8.68               | shell       |
| K22F-ARM | K22F          | mbed-os-core-tests-mbedmicro-mbed-div                                                     | OK      | 9.22               | shell       |
| K22F-ARM | K22F          | mbed-os-core-tests-mbedmicro-rtos-mbed-basic                                              | OK      | 23.41              | shell       |
| K22F-ARM | K22F          | mbed-os-core-tests-mbedmicro-rtos-mbed-isr                                                | TIMEOUT | 67.03              | shell       |
| K22F-ARM | K22F          | mbed-os-core-tests-mbedmicro-rtos-mbed-mail                                               | OK      | 10.25              | shell       |
| K22F-ARM | K22F          | mbed-os-core-tests-mbedmicro-rtos-mbed-mutex                                              | OK      | 18.75              | shell       |
| K22F-ARM | K22F          | mbed-os-core-tests-mbedmicro-rtos-mbed-queue                                              | OK      | 10.28              | shell       |
| K22F-ARM | K22F          | mbed-os-core-tests-mbedmicro-rtos-mbed-semaphore                                          | OK      | 15.97              | shell       |
| K22F-ARM | K22F          | mbed-os-core-tests-mbedmicro-rtos-mbed-signals                                            | TIMEOUT | 66.99              | shell       |
| K22F-ARM | K22F          | mbed-os-core-tests-mbedmicro-rtos-mbed-threads                                            | TIMEOUT | 67.18              | shell       |
| K22F-ARM | K22F          | mbed-os-core-tests-mbedmicro-rtos-mbed-timer                                              | OK      | 23.52              | shell       |
+----------+---------------+-------------------------------------------------------------------------------------------+---------+--------------------+-------------+
mbedgt: test suite results: 31 OK / 8 TIMEOUT
mbedgt: test case report:
+----------+---------------+-------------------------------------------------------------------------------------------+--------------------------------------------------------+--------+--------+--------+--------------------+
| target   | platform_name | test suite                                                                                | test case                                              | passed | failed | result | elapsed_time (sec) |
+----------+---------------+-------------------------------------------------------------------------------------------+--------------------------------------------------------+--------+--------+--------+--------------------+
| K22F-ARM | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-basic_test                        | Repeating Test                                         | 2      | 0      | OK     | 0.07               |
| K22F-ARM | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-basic_test                        | Simple Test                                            | 1      | 0      | OK     | 0.04               |
[... cut ...]
+----------+---------------+-------------------------------------------------------------------------------------------+--------------------------------------------------------+--------+--------+--------+--------------------+
mbedgt: test case results: 132 OK / 2 ERROR
mbedgt: completed in 966.55 sec
mbedgt: exited with code 8
[mbed] ERROR: "mbedgt" returned error code 8.
[mbed] ERROR: Command "mbedgt --test-spec C:\Users\toywat01\Documents\GitHub\oob2\.build/tests\K22F\ARM\test_spec.json" in "C:\Users\toywat01\Documents\GitHub\oob2"
---

```

bridadan, 2016-07-29:
@toyowata Thanks for reporting this. I have a CI run of these tests from PR #2305 where all the tests are passing. Without seeing the verbose test results it's hard to say what the failures were. It also looks like there's a mismatch between the top report's `TIMEOUT`s and the bottom reports `ERROR`s. Do you have the full log of this by any chance?

Also, do you know what version of DAPLink you're running on the K22F? If you run `mbedls` it should show that information in the table.

Thanks!

sg-, 2016-07-29:
What version of interface firmware do you have? https://github.com/mbedmicro/DAPLink/releases/tag/v0241

Also there may be older requirements that need an update??
```
pip install -U -r requirements.txt
```

toyowata, 2016-08-01:
@bridadan @sg- I have updated the DAPLink firmware and then I could not get any TIMEOUT error at all.  Thank you very much.  Closed this ticket.

closed 2016-08-01 by toyowata