ARMmbed/mbed-os #2307: K22F tests fail - GCC OoB RC#3

toyowata, 2016-07-29:
Version: https://github.com/mbedmicro/mbed/releases/tag/mbed-os-5.1.0-rc3
GCC toolchain
Target: K22F
Host OS: Windows 7

```

mbedgt: test suite report:
+--------------+---------------+-------------------------------------------------------------------------------------------+---------+--------------------+-------------+
| target       | platform_name | test suite                                                                                | result  | elapsed_time (sec) | copy_method |
+--------------+---------------+-------------------------------------------------------------------------------------------+---------+--------------------+-------------+
| K22F-GCC_ARM | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-basic_test                        | OK      | 10.4               | shell       |
| K22F-GCC_ARM | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-basic_test_default                | OK      | 10.41              | shell       |
| K22F-GCC_ARM | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-case_async_validate               | OK      | 12.29              | shell       |
| K22F-GCC_ARM | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-case_control_async                | TIMEOUT | 68.17              | shell       |
| K22F-GCC_ARM | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-case_control_repeat               | OK      | 12.38              | shell       |
| K22F-GCC_ARM | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-case_selection                    | TIMEOUT | 68.33              | shell       |
| K22F-GCC_ARM | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-case_setup_failure                | OK      | 10.94              | shell       |
| K22F-GCC_ARM | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-case_teardown_failure             | OK      | 10.94              | shell       |
| K22F-GCC_ARM | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-control_type                      | OK      | 11.16              | shell       |
| K22F-GCC_ARM | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-minimal_async_scheduler           | OK      | 11.36              | shell       |
| K22F-GCC_ARM | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-minimal_scheduler                 | OK      | 11.98              | shell       |
| K22F-GCC_ARM | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-test_assertion_failure_test_setup | OK      | 10.08              | shell       |
| K22F-GCC_ARM | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-test_setup_case_selection_failure | TIMEOUT | 68.09              | shell       |
| K22F-GCC_ARM | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-test_setup_failure                | TIMEOUT | 68.22              | shell       |
| K22F-GCC_ARM | K22F          | mbed-os-core-tests-integration-basic                                                      | OK      | 9.56               | shell       |
| K22F-GCC_ARM | K22F          | mbed-os-core-tests-integration-threaded_blinky                                            | OK      | 29.77              | shell       |
| K22F-GCC_ARM | K22F          | mbed-os-core-tests-mbed_drivers-c_strings                                                 | OK      | 12.52              | shell       |
| K22F-GCC_ARM | K22F          | mbed-os-core-tests-mbed_drivers-callback                                                  | TIMEOUT | 68.89              | shell       |
| K22F-GCC_ARM | K22F          | mbed-os-core-tests-mbed_drivers-dev_null                                                  | OK      | 12.4               | shell       |
| K22F-GCC_ARM | K22F          | mbed-os-core-tests-mbed_drivers-echo                                                      | OK      | 21.3               | shell       |
| K22F-GCC_ARM | K22F          | mbed-os-core-tests-mbed_drivers-generic_tests                                             | OK      | 10.89              | shell       |
| K22F-GCC_ARM | K22F          | mbed-os-core-tests-mbed_drivers-rtc                                                       | TIMEOUT | 68.72              | shell       |
| K22F-GCC_ARM | K22F          | mbed-os-core-tests-mbed_drivers-stl_features                                              | TIMEOUT | 68.74              | shell       |
| K22F-GCC_ARM | K22F          | mbed-os-core-tests-mbed_drivers-ticker                                                    | OK      | 41.44              | shell       |
| K22F-GCC_ARM | K22F          | mbed-os-core-tests-mbed_drivers-timeout                                                   | TIMEOUT | 68.1               | shell       |
| K22F-GCC_ARM | K22F          | mbed-os-core-tests-mbed_drivers-wait_us                                                   | OK      | 20.13              | shell       |
| K22F-GCC_ARM | K22F          | mbed-os-core-tests-mbedmicro-mbed-attributes                                              | OK      | 12.45              | shell       |
| K22F-GCC_ARM | K22F          | mbed-os-core-tests-mbedmicro-mbed-call_before_main                                        | OK      | 9.78               | shell       |
| K22F-GCC_ARM | K22F          | mbed-os-core-tests-mbedmicro-mbed-cpp                                                     | OK      | 9.91               | shell       |
| K22F-GCC_ARM | K22F          | mbed-os-core-tests-mbedmicro-mbed-div                                                     | TIMEOUT | 68.16              | shell       |
| K22F-GCC_ARM | K22F          | mbed-os-core-tests-mbedmicro-rtos-mbed-basic                                              | OK      | 24.67              | shell       |
| K22F-GCC_ARM | K22F          | mbed-os-core-tests-mbedmicro-rtos-mbed-isr                                                | OK      | 14.86              | shell       |
| K22F-GCC_ARM | K22F          | mbed-os-core-tests-mbedmicro-rtos-mbed-mail                                               | OK      | 11.36              | shell       |
| K22F-GCC_ARM | K22F          | mbed-os-core-tests-mbedmicro-rtos-mbed-mutex                                              | OK      | 20.2               | shell       |
| K22F-GCC_ARM | K22F          | mbed-os-core-tests-mbedmicro-rtos-mbed-queue                                              | OK      | 11.34              | shell       |
| K22F-GCC_ARM | K22F          | mbed-os-core-tests-mbedmicro-rtos-mbed-semaphore                                          | OK      | 17.26              | shell       |
| K22F-GCC_ARM | K22F          | mbed-os-core-tests-mbedmicro-rtos-mbed-signals                                            | OK      | 15.09              | shell       |
| K22F-GCC_ARM | K22F          | mbed-os-core-tests-mbedmicro-rtos-mbed-threads                                            | OK      | 15.58              | shell       |
| K22F-GCC_ARM | K22F          | mbed-os-core-tests-mbedmicro-rtos-mbed-timer                                              | OK      | 24.61              | shell       |
+--------------+---------------+-------------------------------------------------------------------------------------------+---------+--------------------+-------------+
mbedgt: test suite results: 30 OK / 9 TIMEOUT
mbedgt: test case report:
+--------------+---------------+-------------------------------------------------------------------------------------------+--------------------------------------------------------+--------+--------+--------+--------------------+
| target       | platform_name | test suite                                                                                | test case                                              | passed | failed | result | elapsed_time (sec) |
+--------------+---------------+-------------------------------------------------------------------------------------------+--------------------------------------------------------+--------+--------+--------+--------------------+
| K22F-GCC_ARM | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-basic_test                        | Repeating Test                                         | 2      | 0      | OK     | 0.08               |
[... cut ...]
+--------------+---------------+-------------------------------------------------------------------------------------------+--------------------------------------------------------+--------+--------+--------+--------------------+
mbedgt: test case results: 133 OK / 1 ERROR
mbedgt: completed in 1073.40 sec
mbedgt: exited with code 9
[mbed] ERROR: "mbedgt" returned error code 9.
[mbed] ERROR: Command "mbedgt --test-spec C:\Users\toywat01\Documents\GitHub\oob2\.build/tests\K22F\GCC_ARM\test_spec.json" in "C:\Users\toywat01\Documents\GitHub\oob2"
---

```

sg-, 2016-07-29:
What version of interface firmware do you have? https://github.com/mbedmicro/DAPLink/releases/tag/v0241

Also there may be older requirements that need an update??
```
pip install -U -r requirements.txt
```

toyowata, 2016-08-01:
Same as https://github.com/ARMmbed/mbed-os/issues/2306

closed 2016-08-01 by toyowata