ARMmbed/mbed-os #2308: K22F tests fail - IAR OoB RC#3

toyowata, 2016-07-29:
Version: https://github.com/mbedmicro/mbed/releases/tag/mbed-os-5.1.0-rc3
IAR toolchain
Target: K22F
Host OS: Windows 7

```

mbedgt: test suite report:
+----------+---------------+-------------------------------------------------------------------------------------------+---------+--------------------+-------------+
| target   | platform_name | test suite                                                                                | result  | elapsed_time (sec) | copy_method |
+----------+---------------+-------------------------------------------------------------------------------------------+---------+--------------------+-------------+
| K22F-IAR | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-basic_test                        | TIMEOUT | 67.03              | shell       |
| K22F-IAR | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-basic_test_default                | OK      | 9.27               | shell       |
| K22F-IAR | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-case_async_validate               | OK      | 10.94              | shell       |
| K22F-IAR | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-case_control_async                | OK      | 17.25              | shell       |
| K22F-IAR | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-case_control_repeat               | OK      | 11.21              | shell       |
| K22F-IAR | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-case_selection                    | OK      | 9.1                | shell       |
| K22F-IAR | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-case_setup_failure                | OK      | 9.76               | shell       |
| K22F-IAR | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-case_teardown_failure             | OK      | 9.76               | shell       |
| K22F-IAR | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-control_type                      | OK      | 10.01              | shell       |
| K22F-IAR | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-minimal_async_scheduler           | OK      | 10.01              | shell       |
| K22F-IAR | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-minimal_scheduler                 | OK      | 10.74              | shell       |
| K22F-IAR | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-test_assertion_failure_test_setup | TIMEOUT | 66.92              | shell       |
| K22F-IAR | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-test_setup_case_selection_failure | TIMEOUT | 66.91              | shell       |
| K22F-IAR | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-test_setup_failure                | TIMEOUT | 66.9               | shell       |
| K22F-IAR | K22F          | mbed-os-core-tests-integration-basic                                                      | OK      | 8.36               | shell       |
| K22F-IAR | K22F          | mbed-os-core-tests-integration-threaded_blinky                                            | OK      | 28.45              | shell       |
| K22F-IAR | K22F          | mbed-os-core-tests-mbed_drivers-c_strings                                                 | OK      | 11.06              | shell       |
| K22F-IAR | K22F          | mbed-os-core-tests-mbed_drivers-callback                                                  | TIMEOUT | 67.78              | shell       |
| K22F-IAR | K22F          | mbed-os-core-tests-mbed_drivers-dev_null                                                  | OK      | 10.74              | shell       |
| K22F-IAR | K22F          | mbed-os-core-tests-mbed_drivers-echo                                                      | OK      | 20.21              | shell       |
| K22F-IAR | K22F          | mbed-os-core-tests-mbed_drivers-generic_tests                                             | OK      | 9.61               | shell       |
| K22F-IAR | K22F          | mbed-os-core-tests-mbed_drivers-rtc                                                       | TIMEOUT | 67.1               | shell       |
| K22F-IAR | K22F          | mbed-os-core-tests-mbed_drivers-stl_features                                              | OK      | 9.82               | shell       |
| K22F-IAR | K22F          | mbed-os-core-tests-mbed_drivers-ticker                                                    | OK      | 40.3               | shell       |
| K22F-IAR | K22F          | mbed-os-core-tests-mbed_drivers-timeout                                                   | OK      | 20.2               | shell       |
| K22F-IAR | K22F          | mbed-os-core-tests-mbed_drivers-wait_us                                                   | OK      | 18.88              | shell       |
| K22F-IAR | K22F          | mbed-os-core-tests-mbedmicro-mbed-attributes                                              | OK      | 10.97              | shell       |
| K22F-IAR | K22F          | mbed-os-core-tests-mbedmicro-mbed-call_before_main                                        | OK      | 8.4                | shell       |
| K22F-IAR | K22F          | mbed-os-core-tests-mbedmicro-mbed-cpp                                                     | OK      | 8.54               | shell       |
| K22F-IAR | K22F          | mbed-os-core-tests-mbedmicro-mbed-div                                                     | TIMEOUT | 66.87              | shell       |
| K22F-IAR | K22F          | mbed-os-core-tests-mbedmicro-rtos-mbed-basic                                              | OK      | 23.36              | shell       |
| K22F-IAR | K22F          | mbed-os-core-tests-mbedmicro-rtos-mbed-isr                                                | TIMEOUT | 66.85              | shell       |
| K22F-IAR | K22F          | mbed-os-core-tests-mbedmicro-rtos-mbed-mail                                               | OK      | 10.13              | shell       |
| K22F-IAR | K22F          | mbed-os-core-tests-mbedmicro-rtos-mbed-mutex                                              | OK      | 18.7               | shell       |
| K22F-IAR | K22F          | mbed-os-core-tests-mbedmicro-rtos-mbed-queue                                              | OK      | 10.16              | shell       |
| K22F-IAR | K22F          | mbed-os-core-tests-mbedmicro-rtos-mbed-semaphore                                          | OK      | 15.95              | shell       |
| K22F-IAR | K22F          | mbed-os-core-tests-mbedmicro-rtos-mbed-signals                                            | OK      | 13.52              | shell       |
| K22F-IAR | K22F          | mbed-os-core-tests-mbedmicro-rtos-mbed-threads                                            | TIMEOUT | 67.1               | shell       |
| K22F-IAR | K22F          | mbed-os-core-tests-mbedmicro-rtos-mbed-timer                                              | OK      | 23.34              | shell       |
+----------+---------------+-------------------------------------------------------------------------------------------+---------+--------------------+-------------+
mbedgt: test suite results: 30 OK / 9 TIMEOUT
mbedgt: test case report:
+----------+---------------+-------------------------------------------------------------------------------------------+--------------------------------------------------------+--------+--------+--------+--------------------+
| target   | platform_name | test suite                                                                                | test case                                              | passed | failed | result | elapsed_time (sec) |
+----------+---------------+-------------------------------------------------------------------------------------------+--------------------------------------------------------+--------+--------+--------+--------------------+
| K22F-IAR | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-basic_test                        | Repeating Test                                         | 2      | 0      | OK     | 0.08               |
| K22F-IAR | K22F          | mbed-os-core-features-frameworks-utest-tests-unit_tests-basic_test                        | Simple Test                                            | 1      | 0      | OK     | 0.04               |
[... cut ...]
+----------+---------------+-------------------------------------------------------------------------------------------+--------------------------------------------------------+--------+--------+--------+--------------------+
mbedgt: test case results: 132 OK / 2 ERROR
mbedgt: completed in 1033.56 sec
mbedgt: exited with code 9
[mbed] ERROR: "mbedgt" returned error code 9.
[mbed] ERROR: Command "mbedgt --test-spec C:\Users\toywat01\Documents\GitHub\oob2\.build/tests\K22F\IAR\test_spec.json" in "C:\Users\toywat01\Documents\GitHub\oob2"
---

```

sg-, 2016-07-29:
What version of interface firmware do you have? https://github.com/mbedmicro/DAPLink/releases/tag/v0241

toyowata, 2016-08-01:
Same as https://github.com/ARMmbed/mbed-os/issues/2306

closed 2016-08-01 by toyowata