ARMmbed/mbed-os #2311: libstdc++ locale-inst.o

MarceloSalazar, 2016-07-29:
Note: ticket originally raised here: https://github.com/ARMmbed/mbed-os/issues/484

### Description
- Type: Question
- Priority: Blocker

## Question

**What pulls in locale-inst.o and wlocale-inst.o?**

When investigating, why mbedOS is so big ROM-wise; i modified memap.py to show deeper into the directory structure which components are big. This led to the following output:

```
+---------------------------------------------------------------------------+--------+-------+-------+
| Module                                                                    |  .text | .data |  .bss |
+---------------------------------------------------------------------------+--------+-------+-------+
...
| libstdc++.a(locale-inst.o)                                                |  39740 |     4 |    96 |
| libstdc++.a(wlocale-inst.o)                                               |  39689 |     4 |    96 |
...
```

What is pulling in 80k of locale support on an embedded device? Application is mbed-os-example-client.

@MarceloSalazar
@markus-becker-tridonic-com

yogpan01, 2016-07-29:
The usage of stringstream in example application was pulling lot of extra code, the application is since then modified to remove the usage of stringstream. Latest updated application should be smaller in code size.
@MarceloSalazar

0xc0170, 2016-07-29:
> What pulls in locale-inst.o and wlocale-inst.o?

That's a good question. Can you please verify another apps that locales are not pulled in by default by mbed. I dont recall they are. Therefore this would help to pinpoint which module requires locales. 

I had a quick look at sources, no ``locale`` include, but ctype yes, here : https://github.com/mbedmicro/mbed/blob/5d7317626c1ad0d1ea37af7c77e7b541fc00f54d/features/FEATURE_CLIENT/mbed-client/source/m2mbase.cpp#L24
cc @yogpan01

0xc0170, 2016-08-09:
Any update to this issue?

yogpan01, 2016-08-09:
This issue should be now fixed, example client was using stringstream for some string operations, which was pulling in the std::locale, that has been removed and hence you should not see the locale being part of the ROM code size anymore. 
Can you please confirm and close this issue ?

0xc0170, 2016-08-09:
@MarceloSalazar @markus-becker-tridonic-com please verify

MarceloSalazar, 2016-08-19:
I understand this is fixed. Please reopen if it's not correct.

closed 2016-08-19 by MarceloSalazar