ARMmbed/mbed-os #2313: export mbed-blinky to uvision4 and open it in uvison5, got compile error

sunsmilearm, 2016-07-29:
linking...
.\.build\uvision4\Unnamed_Project.axf: error: L6002U: Could not open file .\.build\uvision4\softdevice_handler_appsh.o: No such file or directory
Finished: 0 information, 0 warning, 0 error and 1 fatal error messages.
".\.build\uvision4\Unnamed_Project.axf" - 1 Error(s), 25 Warning(s).
Target not created.

0xc0170, 2016-07-29:
please provide full cmd, to reproduce. I'll try to do it locally

cc @pan-

0xc0170, 2016-07-29:
i can't reproduce this. using the same command (I got currently latest mbed master for testing), and exported project builds without errors.

Please provide additional details for us to investigate. Path to a project? How long is it?

bogdanm, 2016-08-02:
Same here: I can't reproduce with uVision 5.20 (using mbed-os RC4). @sunsmilearm, please provide more reproducibility instructions.

ohagendorf, 2016-08-02:
I've already had very strange behaviour, exporting something to uvision4 and using the project in uvision5. I think, the conversion procedure in uvision5 when opening old project files has some problems. But the uvision5 exporter is ready to use. So there is no reason to use uvision4 exported projects in uvision5.

0xc0170, 2016-08-09:
@sunsmilearm Without further information how to reproduce, we will close this issue.

sg-, 2016-09-22:
Closing all older exporter related issues. Please reopen if still valid. https://developer.mbed.org/blog/entry/Fixed-exporters-in-the-mbed-Online-Compi/

closed 2016-09-22 by sg-