ARMmbed/mbed-os #2320: [tools] Exporting mbed-os-example-uvisor fails to run

geky, 2016-07-29:
Original issue https://github.com/ARMmbed/mbed-os-example-uvisor/issues/8:

It looks like the culprit is this line, which specifically removes trailing `F`s from targets if uvisor is enabled:
https://github.com/mbedmicro/mbed/commit/c2e3001739cd91e79c3d3874678a8714b0c075d1#diff-486f048c4c5775670840b518bf8b0ad5R243

This is not playing nicely with the exporters, and may not play nicely with other components of the build system.

cc @screamerbg, @theotherjimmy, @sg-

AlessandroA, 2016-08-02:
I agree that playing with the core name in the build system is not the right way to do it.

In uVisor we disable native FPU instructions because they change the exception behavior in a way which we do not support yet.

It would be useful to have a build option or configuration override to switch between "with" and "without native FPU instructions" (`soft` vs `softfp`/`hardfp` in GCC).

The build system would check that the configuration makes sense (`softfp` and `hardfp` would not be allowed on devices without an FPU).

0xc0170, 2016-08-09:
> I agree that playing with the core name in the build system is not the right way to do it.

+1

>It would be useful to have a build option or configuration override to switch between "with" and "without native FPU instructions" (soft vs softfp/hardfp in GCC).

We had ``fpu`` as part of the target configuration, that could be ``none``, ``single`` or ``double``. but was reverted as someone objected it was too big change for something that could be solved via target suffixes. Does ``soft/fp or hardfp`` make more sense? I dont recall those options are available for tools like uvision and IAR (most of them provide either enabled or disabled, because of cortex-m7, they changed to none, single, double).

AlessandroA, 2016-08-10:
I think that single vs. double FPU is orthogonal to this. The switch I'm proposing allows to choose between "hardware FPU instructions" and "software-emulated FPU instructions". All toolchains should support this out of the box.

Just for reference: Both `softfp` vs `hardfp` imply the use of hardware FPU instructions. The only difference is in the resulting ABI conventions:

> Specifying ‘soft’ causes GCC to generate output containing library calls for floating-point operations. ‘softfp’ allows the generation of code using hardware floating-point instructions, but still uses the soft-float calling conventions. ‘hard’ allows generation of floating-point instructions and uses FPU-specific calling conventions.

If this additional choice is only available to GCC we can ignore it.

I think there is a value in using sw FP instructions even when the hardware has an FPU, for example to reduce interrupt latency, or to make it more deterministic (in the case of lazy FPU stacking).

@0xc0170

bogdanm, 2016-08-10:
While that's true, we need to take into consideration the implications that such a change would have on our binary releases (SDK, nanostack...). It's not impossible to do, but it might be a bit more difficult than it appears to be.

AlessandroA, 2016-08-10:
Yes definitely, but that makes the point even more important. If a binary lib has been compiled with `softfp`, then it uses the hw FPU instructions but relies on a soft-FPU ABI, which would make the calls to that lib unpredictably buggy in case the rest of the app is compiled with `hardfp`. Are we currently making sure that these modules are built in a consistent way?

theotherjimmy, 2016-08-11:
At least for GCC, the linker will simply refuse to link if the ABIs are different.

sg-, 2016-09-22:
Will look at making target.core a parameter that can be set from the application. that should remove the magic stripping in the gcc.py file (or __init__ whichever) and allow the override to also be passed to exporters.

theotherjimmy, 2016-10-20:
Unfortunately this is still a thing.

theotherjimmy, 2017-04-03:
Looks to be exactly the same for me. Closing.

closed 2017-04-03 by theotherjimmy