ARMmbed/mbed-os #2324: localtime() broken with GCC ARM toolchain

ccli8, 2016-08-01:
***mbed-os sha***
4b506280eb4d34b3ed0e5aa82600d478e8f4ea1c
***mbed-cli version***
0.8.8
***GCC ARM version***
4.9.3
***Target board***
K64F
***Steps to Reproduce***
Program gets broken in localtime(). Without __disable_irq(), it is fine.
```
time_t t = 1256729737;
__disable_irq();
struct tm *timeinfo = localtime(&t);
```

0xc0170, 2016-08-09:
>  Program gets broken in localtime(). Without __disable_irq(), it is fine.

Does that mean that you workaround the issue by using ``__disable_irq`` ? I would recommend using ``core_util_critical_section_enter/exit``. Apart from it, what does it mean broken?

I would like to know what we should reproduce . The output without disabled irq and with disabled irq.  any other toolchain is ok , only gcc arm ?

ccli8, 2016-08-10:
@0xc0170 With gcc arm toolchain, you can reproduce the issue by just running the code snippet. Program will trap in ``localtime`` with ``__disable_irq`` in front. With arm toolchain instead, it is OK.
```
time_t t = 1256729737;
__disable_irq();
struct tm *timeinfo = localtime(&t);
```
Substituting ``core_util_critical_section_enter`` for ``__disable_irq`` doesn't help because ``core_util_critical_section_enter`` calls ``__disable_irq`` and program traps as above. The preceding ``__disable_irq`` is intentional to reproduce the issue.

The issue originally comes from test code mbed-os/TESTS/mbed_drivers/rtc in my NUMAKER_PFM_NUC472 porting on GCC_ARM toolchain. In my implementation of rtc hal, I call ``localtime`` for conversion between calendar time and timestamp. I just reduce the issue on K64F with code snippet above.

ccli8, 2016-08-11:
The issue was launched to address test failure of test code `mbed-os/TESTS/mbed_drivers/rtc` (mbed OS 5.0.1) in my NUMAKER_PFM_NUC472 porting on GCC_ARM toolchain. Now on mbed OS 5.1, the test is passed due to `set_time` `(mbed/hal/common/mbed_critical.c)` implemented in mutex rather than `core_util_critical_section_enter/exit(__disable/enable_irq)`.

Program still traps in `localtime` with gcc toolchain. It would be related to implementation of `localtime` in gcc.
```
time_t t = 1256729737;
__disable_irq();
struct tm *timeinfo = localtime(&t);
```

The issue is closed due to my original problem fixed.

closed 2016-08-11 by ccli8