ARMmbed/mbed-os #2325: [exporter] FPU option not set in exported IAR project files for several platforms

maclobdell, 2016-08-01:
This is a duplicate of an issue filed in project generator definitions.
https://github.com/project-generator/project_generator_definitions/issues/100

I'm using:
IAR EWARM 7.40.7
mbed-os-5.1.0-rc4
project-generator (0.9.8)
project-generator-definitions (0.2.27)

Exporting to the following targets and trying to build in IAR IDE results in an error that the device doesn't have VFP registers. The project files need to have the FPU option set to VFPv4 single precision.

NUCLEO_F411RE
NUCLEO_F401RE
DISCO_F429ZI
FRDM-K64F
FRDM-K22F

sg-, 2016-08-04:
@0xc0170 @bogdanm I thought 7.4 was suppose to work with our existing template?

bogdanm, 2016-08-04:
That was my understanding too, yes ...

0xc0170, 2016-08-08:
> I thought 7.4 was suppose to work with our existing template?

It shall, I tested K22F 10 days ago and FPU was properly set. I'll continue investigating , I believe @sarahmarshy already looked at it, as she mentioned it in the other issue, I was not aware this is still an issue.

0xc0170, 2016-08-08:
I got 7.40.2 and its working for K22 and NUCLEO_F411RE. I am testing it with project.py for MBED_10. hard to believe that between 7.40.2 and 7.40.7 there was a change for FPU.

0xc0170, 2016-08-08:
I installed 7.40.7 and as stated in another issue about FPU, seems like even lower version (7.40.7) contains a breaking change for FPU. When a target uses FPU2 (new FPU settings). From the release notes, it seems that from version 7.40.2 up it's FPU2 (cortex m7 FPU double precision addition). I don't understand their versioning to include this type of update within a patch version.

I think it's time to update to >7.50, deprecate old FPU settings (update mbed IAR template to newer versions of IAR), and state a specific version of IAR (7.x.x) that we are using and testing against.

@sarahmarshy Any more info for this issue?

sarahmarshy, 2016-08-09:
@0xc0170 I have version 7.40.3, and I was able to replicate this issue. I will see if the new FPU settings are backwards compatible.

0xc0170, 2016-08-10:
> @0xc0170 I have version 7.40.3, and I was able to replicate this issue. I will see if the new FPU settings are backwards compatible.

IAR is not backward compatible, that's not their intention. The new project does not open in the older versions, only forward compatible. That's my experience. Thus FPU2 (new FPU settings) won't open in the older versions (error shown about a project created by a newer version of IAR, something like that). Many projects when they release their code, they ship with updated projects for newer tools (to get new MCU, new settings) and they state the versions they release for so users are aware.

We do not know what users is using. progen could do the transformation, but how to get a version and find out which version was breaking. Therefore this is left to a user (in this case mbed). mbed decides what tools version are used, and use the template from that version. progen gets the template, and uses it. In this case, it would find out that FPU2 is present, thus MCU fpu settings are set there, not into FPU. That's how it is implemented at the moment.

sg-, 2017-01-16:
Exporters have historically worked with the latest version. We'll introduce better specified tested versions of tools with mbed OS 5.4

closed 2017-01-16 by sg-