ARMmbed/mbed-os #2326: Include paths

mottigondabi, 2016-08-01:
The ability to define a specific file for the `mbed compile --source <path-to-file>/<file-name>.c` exist *but* apparently there is no way to define a specific headers only folder (*.h) that will suppress *.c or *.cpp files assuming we would like to avoid using `.mbedignore` for this matter.

i.e.: `mbed compile --source main.cpp --source_incs some_path_to_folder/inner_folder`

The `--source_incs` directive (if was exist) suppresses anything that is not `*.h`

@jenia81 
@screamerbg

0xc0170, 2016-08-09:
I would propose ``--includes`` , as we got ``--sources``, and includes could be either a path or a file.

bogdanm, 2016-08-09:
I'm not sure why we aren't considering a `--source` directory to also be an include directory. This would be consistent with the way our scanner works.

bridadan, 2016-08-09:
It sounds like the feature request here is to be able to include a folder but only allow certain types of files to be included? Maybe instead of creating a `--include`, we could just add globing support to `--source`? So something like `--source=myproj/source/*.h` ?

theotherjimmy, 2016-08-09:
@bogdanm We are. This is a feature request to have a `--source` like argument that only searches for includes.

bridadan, 2016-08-09:
@mottigondabi Is there a reason the source is organized in this way? It seems like a very unusual use case. It seems like the quickest way forward would be to just move the `.h` files into a separate folder and include that.

bogdanm, 2016-08-09:
> This is a feature request to have a --source like argument that only searches for includes.

Sorry, I've apparently mistaken this with another issue.

sima-gp, 2016-09-28:
Hi all, hi @bridadan, hi @theotherjimmy , 

As a prove of concept, I want to build (using tools/build.py) a small application using code external to mbed. To be able to quickly build this application without copying source code, I would like to include header files using the GCC -I flag.
 
When diving into the mbed python code I noticed the following:
- tools/build_api.py: the build_library function provides the functionality to give along include directories (parameter: ‘inc_dirs’ - additional directories where include files may be found)
- tools/build.py is using tis build_library function but without the ‘inc_dirs’ parameter. With other words, this functionality is not exposed to the CLI.
 
Now my question is: What is the ‘mbed way of working’ to include external header files? Is there a configuration file which I’m missing?
Some notes:
 
•        Do I need to add definitions to tools/settings.py or tools/private_settings.py? 
•        I noticed that there is a file automatically generated which has some (‘default’) -Ixxxxxx parameters, “./.build/mbed/.temp/TARGET_Qorvo/TOOLCHAIN_ARM_STD/.includes_c08fabc2ca06f19fcd6a2883245a3dae.txt”. 
It’s clear to me that the build_library function eventually fills in its parameters into this file, but can you point me out in code where the ‘default’ values are written to this file? 

I hope you guys can help me out?

Thanks in advance,
Kr,
Simon

bridadan, 2016-09-28:
@sima-gp You bring up an interesting point. The most logical reason I can think of why an equivalent `-I` option isn't available from the `build.py` CLI is that it was never needed! `build.py` historically knew how to build all the libraries (aka, knew which directories to include/not include). It didn't rely on input from the command line, but rather data stored in the python tools (if you're curious, `libraries.py` is where this data is kept). It was never intended to be a general purpose build system, so this kind of a general flag was never added.

The "mbed way" in the past has always been to just copy the source into your working directory and let the tools do the rest. The 'default' values for the include file you mentioned is built up from the source in your working directory.

theotherjimmy, 2016-09-28:
> What is the ‘mbed way of working’ to include external header files?

At the moment that works as follows:
 1. Ensure that the header files are all in a directory with no source files
 2. specify that directory as a parameter to a `--source` flag

> Is there a configuration file which I’m missing?

There is not, unless you consider `tools/libraries.py` a 'configuration file'.

closed 2017-10-27 by ChiefBureaucraticOfficer

jorisa, 2017-12-21:
I still think there is a case for this. When you submodule other projects there sometimes are files you do not want to include in the build (like test.cpp that also has a main function). You can explicitly add the source files, but there is no way to point it to the headers.

I am aware you could exclude files using mbedignore, but that means you would have to fork the repo.

theotherjimmy, 2017-12-21:
@jorisa `.mbedignore` files work across repo boundaries.

That is:
```
.
├── README.md
├── main.cpp
├── mbed-os
├── .mbedignore
└── mbed-os.lib
```
The `.mbedignore` here can ignore files within the `mbed-os` directory. This allows me to ignore things in sub-repos without a fork of the sub-repo.