ARMmbed/mbed-os #2327: NUCLEO_F303RE (maybe others) SRAM wrong size in keil startup_*****.s

giulianodianda, 2016-08-01:
Hi,
I often run out of memory when mallocating stuff, all the times i fixed the correct SRAM size (btw online compiler doesn't allow to edit the .s, needs to delete and import a fixed one), maybe it's time to open an issue:

https://github.com/ARMmbed/mbed-os/blob/master/hal/targets/cmsis/TARGET_STM/TARGET_STM32F3/TARGET_NUCLEO_F303RE/TOOLCHAIN_ARM_MICRO/startup_stm32f303xe.S#L54

https://github.com/ARMmbed/mbed-os/blob/master/hal/targets/cmsis/TARGET_STM/TARGET_STM32F3/TARGET_NUCLEO_F303RE/TOOLCHAIN_ARM_STD/startup_stm32f303xe.S#L42

16KB size is for CCRAM, but it's mapped at 0x10000000 (probably not used at all in keil, not sure)
64KB is correct size for SRAM, mapped at 0x20000000

Don't know for other Nucleo targets, maybe worths a check.

svastm, 2016-08-03:
Hello,
Thanks for reporting this issue and the fix. I will check the other targets and open a new issue if I find something.

sg-, 2016-08-04:
Thanks for the note about online IDE - this will be fixed 👍

bcostm, 2016-09-09:
This board has been fixed. Can you please close the issue ?

closed 2017-01-16 by sg-