ARMmbed/mbed-os #2335: Intended project structure

javier-moreno-tridonic-com, 2016-08-02:
Hi there, 

We are doing our first attempt to port our current software to mbed5. The way we understood it, the new configuration would allow us to configure a new target per final product that could inherit from the platform target.

The initial idea was to translate all our old private yotta modules with different functionalities into "FEATURE_" folders. This way we could select the appropriate features for our different targets in the targets.json. Allowing us to fit everything (mbed5 + a bunch of own features) in one custom repository (reducing the overhead of maintaining and releasing).

For that approach to work, I had to modify the following code in /mbed-os/tools/config.py 

```python
 # Allowed features in configurations
    __allowed_features = [
        "UVISOR", "BLE", "CLIENT", "IPV4", "IPV6", "COMMON_PAL", "STORAGE", "LOGGER"
    ]
```

Where "FEATURE_LOGGER" is just a dummy feature that I'm using to test this concept. The fact that I needed to do such change makes me think that this is not the intended way to add custom functionality to mbed5. What is then the intended project structure?

@MarceloSalazar @markus-becker-tridonic-com

javier-moreno-tridonic-com, 2016-08-02:
Also could be interesting that features define macros (e.g. #define MBED_FEATURE_IPV4) this way we could differentiate with #ifdef directives when necessary.

markus-becker-tridonic-com, 2016-08-09:
@marcelosalazar

MarceloSalazar, 2016-08-09:
@bogdanm could you please have a look to this?

MarceloSalazar, 2016-08-16:
@sg- @screamerbg could you please comment on this question?

sg-, 2016-08-17:
Features are not intended to be used by anything outside the mbed OS tree.

javier-moreno-tridonic-com, 2016-08-17:
So is there a different way include different code modules based on the target? If not, is there anyone planned?

Is something that we would need and IMHO other people might find quite useful.

sg-, 2016-08-17:
For now: Including code based on target has been done by scanning for extra_labels (TARGET_xxx). If a target emits labels, you can name folders to conditional include based on which is selected. Beware: as SDKs change or target code gets refactored these ```labels``` have also changed breaking people who rely on this. This doesn't happen often, I can site one or three occurrences over the last 3 years. We're working towards normalize this and better document the contract between the codebase and tools.

markus-becker-tridonic-com, 2016-08-18:
@javier-moreno-tridonic-com this works for us.

closed 2016-08-18 by sg-