ARMmbed/mbed-os #2336: RZ_A1H binaries generated on Linux and OSX don't work mbed-os-5.1.0-rc4

bulislaw, 2016-08-02:
Board: RZ_A1H
Version: mbed-os-5.1.0-rc4
OS: Linux/OSX
Toolchain: GCC and ARM

I've tried multiple examples, they compile just fine, but when I flash them, the board doesn't do anything. It seem that they are actually flashed (LED is blinking), then the board reboots, but it doesn't seem to be doing anything (depending on example e.g. no serial output or user led blinking), but the power LED is on and it flashes USB activity every now and again.
At the same time the HelloWorld_blinky.bin demo example on the board mbed page works fine (https://developer.mbed.org/platforms/VK-RZA1H/).

bridadan, 2016-08-02:
@bulislaw If you take the binary that you compiled on Linux/OSX and flash them on a Windows machine, does that work correctly? It's possible that there was an issue during the flashing process even though things appeared to work correctly.

bulislaw, 2016-08-02:
@bridadan I don't have windows so hard to try that. But when I take the binary from (https://developer.mbed.org/platforms/VK-RZA1H/) and flash it from Mac/Linux it works fine.

bridadan, 2016-08-02:
Ah I see. It is possible that the size of the binary is causing issues when flashing. I've seen this before on Mac at least with older firmwares. That's why I was curious to see if the built binary you have works when flashed on Windows for that platform. If the `VK-RZ/A1H` is like the `RZ_A1H`, then it uses CMSIS-DAP and not DAPLink. CMSIS-DAP was most tested on Windows, so its possible this is the issue.

bulislaw, 2016-08-02:
I'm quite sure it's the same, I have both designation printed on the board.

bridadan, 2016-08-02:
Oh sorry, I think what I said previously was confusing. I'm sure you have the right board! I'm just wondering what version of interface firmware the board uses. I know the `RZ_A1H` (GR-PEACH) platform uses CMSIS-DAP, which is a fairly old release of the firmware that can have issues on Linux and Mac. I don't know if the `VK-RZ/A1H` board you have uses a similar version of the firmware, but if it does it could be susceptible to the same issues. This is why I suggested trying to flash the built binary on Windows to try and rule out the actual binary as the failure point.

toyowata, 2016-08-03:
```VK-RZ/A1H``` has same chip as ```RZ_A1H```, but different memory configuration.  The binary for the ```VK-RZ/A1H``` should not work on the ```RZ_A1H``` (GR-PEACH).

bulislaw, 2016-08-03:
Ok, my bad. It seems that my board has both `VK-RZ/A1H` and `RZ_A1H` markings on it, but it only works with the former one as @toyowata mentioned. Closing.

closed 2016-08-03 by bulislaw