ARMmbed/mbed-os #2344: FEATURE_COMMON_PAL duplicate file names

sarahmarshy, 2016-08-02:
I was investigating these two issues:
https://github.com/ARMmbed/mbed-os-example-client/issues/62
https://github.com/ARMmbed/mbed-os-example-client/issues/61

The uvision issue was resolved by eliminating the -D defines in the assembler directives of uvision and creating a header file that is preincluded for all files using armcc. After that issue was resolved, I found uvision was also complaining about competing object files (same as IAR)

One example:
/features/FEATURE_COMMON_PAL/sal-stack-nanostack-eventloop/source/timeout.c
./features/FEATURE_COMMON_PAL/sal-stack-nanostack-eventloop/bare-metal/timeout.c

Now, multiple files will be compiled to timeout.o, resulting in errors. 

@0xc0170 @theotherjimmy @sg- @geky

sarahmarshy, 2016-08-02:
Would it be possible to add the macros to mbed_config.h? It is already preincluded.

sg-, 2016-08-05:
>./features/FEATURE_COMMON_PAL/sal-stack-nanostack-eventloop/bare-metal/timeout.c
> ./hal/common/Timeout.cpp

Can you create a template project for a simple blinky (an example from IAR) and lets try to see if any path is preserved and if so how much.

So create
start.asm
start.c
start.cpp

see if this builds. If so start extending paths until it doesn't. Once we better understand the behavior we can have a chat with IAR

sarahmarshy, 2016-08-05:
These 3 files in the same directory do not build.

Using only two files of the same name for mobility:

Moving one of the files to an entirely separate drive does not build.

0xc0170, 2016-08-09:
As I experienced some tools flatten the build directory, they create object files based on the filename, therefore two or more files with the same filename result in the error. I experienced this with coide and IAR. IAR does not take paths to the equation at all.

adustm, 2016-12-01:
Hello, I am facing the same issue trying to compile the mbed-os-client-example in an exported IAR project (compiling with command line 'mbed compile' is not an issue).
I cannot find the option in IAR to recreate the directory tree in the /Obj directory.
Did someone find a way not to build flately in the /build directory ?
Kind regards
Armelle

theotherjimmy, 2016-12-01:
@adustm Sorry to hear that this is affecting you as well. I have not found that option, but if you do, could you send me a project file with that option turned on? I could enable that option by default then.

0xc0170, 2017-03-07:
Is this issue still valid? I believe we fixed it via mbed-cli, but still persist in the IDE as IAR (the workspace have this limitations that two object files cant have the same name as it flattens output).

adustm, 2017-03-07:
Hello, According to IAR help book about `--output {filename | directory} `

> this option is not available in the IDE

I don't know how this could be fixed without renaming files. Unless anyone has another idea ?

Kind regards

closed 2017-10-27 by ChiefBureaucraticOfficer