ARMmbed/mbed-os #2350: target.public needs rename

sg-, 2016-08-03:
This is quite a bit misleading. We should rename this to something more intuitive.

"public" is a bit misleading here. Seems it should be something like "inheritable", "protected", etc.. I'm sure there are OO descriptions that better describe what and how its used.

https://github.com/ARMmbed/mbed-os/pull/2314

0xc0170, 2016-08-09:
@bogdanm Proposals? ``private`` ?

bogdanm, 2016-08-09:
I'd vote for `internal`.

0xc0170, 2016-08-09:
+1, Let's do it.

theotherjimmy, 2016-09-20:
Guys, We can't rename things willy-nilly like this. We have several places (more than just here) that consume the targets.json file and use it. In particular, the website and the mbed-sdk-tools both do this. If you want to make a change like this, please submit PRs too all three locations.

sg-, 2016-09-20:
Just drop links to where the other changes need to also be made.

theotherjimmy, 2016-09-20:
https://developer.mbed.org/users/mbed_official/code/mbed-sdk-tools/ 
https://github.com/ARMmbed/mbed-sdk-web

theotherjimmy, 2016-09-20:
We're working on making these tools follow the mbed-os tools, but that's not done yet.

closed 2017-10-27 by ChiefBureaucraticOfficer