ARMmbed/mbed-os #2360: [IAR Compile]

sarahmarshy, 2016-08-03:
When compiling the [border router](https://github.com/ARMmbed/k64f-border-router) example with IAR: 

```
[OS ERROR] Command: C:/Program Files (x86)/IAR Systems/Embedded Workbench 7.3/arm\bin\iccarm --no_wrap_diagnostics -e --diag_suppress=Pa050,Pa084,Pa093,Pa082 --cpu Cortex-M4F --thumb --dlib_config C:/Program Files (x86)/IAR Systems/Embedded Workbench 7.3/arm\inc\c\DLib_Config_Full.h -Oh --c++ --no_rtti --no_exceptions --guard_calls -DFEATURE_IPV6=1 -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -D__FPU_PRESENT=1 -DTARGET_Freescale -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -DDEVICE_RTC=1 -DTOOLCHAIN_object -D__CMSIS_RTOS -DFSL_RTOS_MBED -DDEVICE_STORAGE=1 -DMBEDTLS_ENTROPY_HARDWARE_ALT -DTARGET_KPSDK_MCUS -DMBED_BUILD_TIMESTAMP=1470256340.89 -DTARGET_CORTEX_M -DTARGET_KSDK2_MCUS -DTARGET_LIKE_CORTEX_M4 -DDEVICE_ANALOGOUT=1 -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_K64F -DTARGET_FF_ARDUINO -DTOOLCHAIN_IAR -DDEVICE_INTERRUPTIN=1 -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -D__CORTEX_M4 -DDEVICE_STDIO_MESSAGES=1 -DCPU_MK64FN1M0VMD12 -DFEATURE_IPV4=1 -DTARGET_LIKE_MBED -DFEATURE_COMMON_PAL=1 -DTARGET_KPSDK_CODE -DTARGET_RELEASE -DDEVICE_SERIAL=1 -DFEATURE_STORAGE=1 -D__MBED_CMSIS_RTOS_CM -DDEVICE_SLEEP=1 -DTARGET_FRDM -DDEVICE_SPI=1 -DDEVICE_ERROR_RED=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DDEVICE_PORTIN=1 -DTARGET_MCU_K64F -DARM_MATH_CM4 -DDEVICE_SERIAL_FC=1 -f .\.build\k64f\IAR\.includes_1ee4b76cbaabfd720b93e333202f9e00.txt --preinclude=.\.build\k64f\IAR\mbed_config.h --dependencies .\.build\k64f\IAR\atmel-rf-driver\source\at24mac.d -l .\.build\k64f\IAR\atmel-rf-driver\source\at24mac.s.txt -o .\.build\k64f\IAR\atmel-rf-driver\source\at24mac.o .\atmel-rf-driver\source\at24mac.cpp
[OS ERROR] Command: C:/Program Files (x86)/IAR Systems/Embedded Workbench 7.3/arm\bin\iccarm --no_wrap_diagnostics -e --diag_suppress=Pa050,Pa084,Pa093,Pa082 --cpu Cortex-M4F --thumb --dlib_config C:/Program Files (x86)/IAR Systems/Embedded Workbench 7.3/arm\inc\c\DLib_Config_Full.h -Oh --vla -DFEATURE_IPV6=1 -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -D__FPU_PRESENT=1 -DTARGET_Freescale -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -DDEVICE_RTC=1 -DTOOLCHAIN_object -D__CMSIS_RTOS -DFSL_RTOS_MBED -DDEVICE_STORAGE=1 -DMBEDTLS_ENTROPY_HARDWARE_ALT -DTARGET_KPSDK_MCUS -DMBED_BUILD_TIMESTAMP=1470256340.89 -DTARGET_CORTEX_M -DTARGET_KSDK2_MCUS -DTARGET_LIKE_CORTEX_M4 -DDEVICE_ANALOGOUT=1 -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_K64F -DTARGET_FF_ARDUINO -DTOOLCHAIN_IAR -DDEVICE_INTERRUPTIN=1 -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -D__CORTEX_M4 -DDEVICE_STDIO_MESSAGES=1 -DCPU_MK64FN1M0VMD12 -DFEATURE_IPV4=1 -DTARGET_LIKE_MBED -DFEATURE_COMMON_PAL=1 -DTARGET_KPSDK_CODE -DTARGET_RELEASE -DDEVICE_SERIAL=1 -DFEATURE_STORAGE=1 -D__MBED_CMSIS_RTOS_CM -DDEVICE_SLEEP=1 -DTARGET_FRDM -DDEVICE_SPI=1 -DDEVICE_ERROR_RED=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DDEVICE_PORTIN=1 -DTARGET_MCU_K64F -DARM_MATH_CM4 -DDEVICE_SERIAL_FC=1 -f .\.build\k64f\IAR\.includes_1ee4b76cbaabfd720b93e333202f9e00.txt --preinclude=.\.build\k64f\IAR\mbed_config.h --dependencies .\.build\k64f\IAR\mbed-os\features\FEATURE_COMMON_PAL\mbed-client-c\source\libCoap\src\sn_coap_header_check.d -l .\.build\k64f\IAR\mbed-os\features\FEATURE_COMMON_PAL\mbed-client-c\source\libCoap\src\sn_coap_header_check.s.txt -o .\.build\k64f\IAR\mbed-os\features\FEATURE_COMMON_PAL\mbed-client-c\source\libCoap\src\sn_coap_header_check.o .\mbed-os\features\FEATURE_COMMON_PAL\mbed-client-c\source\libCoap\src\sn_coap_header_check.c
[OS ERROR] Command: C:/Program Files (x86)/IAR Systems/Embedded Workbench 7.3/arm\bin\iccarm --no_wrap_diagnostics -e --diag_suppress=Pa050,Pa084,Pa093,Pa082 --cpu Cortex-M4F --thumb --dlib_config C:/Program Files (x86)/IAR Systems/Embedded Workbench 7.3/arm\inc\c\DLib_Config_Full.h -Oh --vla -DFEATURE_IPV6=1 -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -D__FPU_PRESENT=1 -DTARGET_Freescale -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -DDEVICE_RTC=1 -DTOOLCHAIN_object -D__CMSIS_RTOS -DFSL_RTOS_MBED -DDEVICE_STORAGE=1 -DMBEDTLS_ENTROPY_HARDWARE_ALT -DTARGET_KPSDK_MCUS -DMBED_BUILD_TIMESTAMP=1470256340.89 -DTARGET_CORTEX_M -DTARGET_KSDK2_MCUS -DTARGET_LIKE_CORTEX_M4 -DDEVICE_ANALOGOUT=1 -DTARGET_M4[OS ERROR] Command: C:/Program Files (x86)/IAR Systems/Embedded Workbench 7.3/arm\bin\iccarm --no_wrap_diagnostics -e --diag_suppress=Pa050,Pa084,Pa093,Pa082 --cpu Cortex-M4F --thumb --dlib_config C:/Program Files (x86)/IAR Systems/Embedded Workbench 7.3/arm\inc\c\DLib_Config_Full.h -Oh --vla -DFEATURE_IPV6=1 -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -D__FPU_PRESENT=1 -DTARGET_Freescale -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -DDEVICE_RTC=1 -DTOOLCHAIN_object -D__CMSIS_RTOS -DFSL_RTOS_MBED -DDEVICE_STORAGE=1 -DMBEDTLS_ENTROPY_HARDWARE_ALT -DTARGET_KPSDK_MCUS -DMBED_BUILD_TIMESTAMP=1470256340.89 -DTARGET_CORTEX_M -DTARGET_KSDK2_MCUS -DTARGET_LIKE_CORTEX_M4 -DDEVICE_ANALOGOUT=1 -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_K64F -DTARGET_FF_ARDUINO -DTOOLCHAIN_IAR -DDEVICE_INTERRUPTIN=1 -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -D__CORTEX_M4 -DDEVICE_STDIO_MESSAGES=1 -DCPU_MK64FN1M0VMD12 -DFEATURE_IPV4=1 -DTARGET_LIKE_MBED -DFEATURE_COMMON_PAL=1 -DTARGET_KPSDK_CODE -DTARGET_RELEASE -DDEVICE_SERIAL=1 -DFEATURE_STORAGE=1 -D__MBED_CMSIS_RTOS_CM -DDEVICE_SLEEP=1 -DTARGET_FRDM -DDEVICE_SPI=1 -DDEVICE_ERROR_RED=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DDEVICE_PORTIN=1 -DTARGET_MCU_K64F -DARM_MATH_CM4 -DDEVICE_SERIAL_FC=1 -f .\.build\k64f\IAR\.includes_1ee4b76cbaabfd720b93e333202f9e00.txt --preinclude=.\.build\k64f\IAR\mbed_config.h --dependencies .\.build\k64f\IAR\atmel-rf-driver\source\driverRFPhy.d -l .\.build\k64f\IAR\atmel-rf-driver\source\driverRFPhy.s.txt -o .\.build\k64f\IAR\atmel-rf-driver\source\driverRFPhy.o .\atmel-rf-driver\source\driverRFPhy.c
[OS ERROR] Command: C:/Program Files (x86)/IAR Systems/Embedded Workbench 7.3/arm\bin\iccarm --no_wrap_diagnostics -e --diag_suppress=Pa050,Pa084,Pa093,Pa082 --cpu Cortex-M4F --thumb --dlib_config C:/Program Files (x86)/IAR Systems/Embedded Workbench 7.3/arm\inc\c\DLib_Config_Full.h -Oh --vla -DFEATURE_IPV6=1 -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -D__FPU_PRESENT=1 -DTARGET_Freescale -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -DDEVICE_RTC=1 -DTOOLCHAIN_object -D__CMSIS_RTOS -DFSL_RTOS_MBED -DDEVICE_STORAGE=1 -DMBEDTLS_ENTROPY_HARDWARE_ALT -DTARGET_KPSDK_MCUS -DMBED_BUILD_TIMESTAMP=1470256340.89 -DTARGET_CORTEX_M -DTARGET_KSDK2_MCUS -DTARGET_LIKE_CORTEX_M4 -DDEVICE_ANALOGOUT=1 -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_K64F -DTARGET_FF_ARDUINO -DTOOLCHAIN_IAR -DDEVICE_INTERRUPTIN=1 -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -D__CORTEX_M4 -DDEVICE_STDIO_MESSAGES=1 -DCPU_MK64FN1M0VMD12 -DFEATURE_IPV4=1 -DTARGET_LIKE_MBED -DFEATURE_COMMON_PAL=1 -DTARGET_KPSDK_CODE -DTARGET_RELEASE -DDEVICE_SERIAL=1 -DFEATURE_STORAGE=1 -D__MBED_CMSIS_RTOS_CM -DDEVICE_SLEEP=1 -DTARGET_FRDM -DDEVICE_SPI=1 -DDEVICE_ERROR_RED=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DDEVICE_PORTIN=1 -DTARGET_MCU_K64F -DARM_MATH_CM4 -DDEVICE_SERIAL_FC=1 -f .\.build\k64f\IAR\.includes_1ee4b76cbaabfd720b93e333202f9e00.txt --preinclude=.\.build\k64f\IAR\mbed_config.h --dependencies .\.build\k64f\IAR\mbed-os\features\FEATURE_COMMON_PAL\mbed-client-c\source\libCoap\src\sn_coap_parser.d -l .\.build\k64f\IAR\mbed-os\features\FEATURE_COMMON_PAL\mbed-client-c\source\libCoap\src\sn_coap_parser.s.txt -o .\.build\k64f\IAR\mbed-os\features\FEATURE_COMMON_PAL\mbed-client-c\source\libCoap\src\sn_coap_parser.o .\mbed-os\features\FEATURE_COMMON_PAL\mbed-client-c\source\libCoap\src\sn_coap_parser.c
[OS ERROR] Command: C:/Program Files (x86)/IAR Systems/Embedded Workbench 7.3/arm\bin\iccarm --no_wrap_diagnostics -e --diag_suppress=Pa050,Pa084,Pa093,Pa082 --cpu Cortex-M4F --thumb --dlib_config C:/Program Files (x86)/IAR Systems/Embedded Workbench 7.3/arm\inc\c\DLib_Config_Full.h -Oh --vla -DFEATURE_IPV6=1 -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -D__FPU_PRESENT=1 -DTARGET_Freescale -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -DDEVICE_RTC=1 -DTOOLCHAIN_object -D__CMSIS_RTOS -DFSL_RTOS_MBED -DDEVICE_STORAGE=1 -DMBEDTLS_ENTROPY_HARDWARE_ALT -DTARGET_KPSDK_MCUS -DMBED_BUILD_TIMESTAMP=1470256340.89 -DTARGET_CORTEX_M -DTARGET_KSDK2_MCUS -DTARGET_LIKE_CORTEX_M4 -DDEVICE_ANALOGOUT=1 -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_K64F -DTARGE[OS ERROR] Command: C:/Program Files (x86)/IAR Systems/Embedded Workbench 7.3/arm\bin\iccarm --no_wrap_diagnostics -e --diag_suppress=Pa050,Pa084,Pa093,Pa082 --cpu Cortex-M4F --thumb --dlib_config C:/Pr [... cut ...] MMON_PAL\nanostack-libservice\source\nvmHelper\ns_nvm_helper.o .\mbed-os\features\FEATURE_COMMON_PAL\nanostack-libservice\source\nvmHelper\ns_nvm_helper.c
[OS ERROR] Command: C:/Program Files (x86)/IAR Systems/Embedded Workbench 7.3/aanostack-eventloop\bare-metal\ns_timer.c
[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u C:\Repos\oob\k64f-border-router\mbed-os\tools\make.py -t IAR -m k64f --source . --build .\.build\k64f\IAR" in "C:\Repos\oob\k64f-border-router
```

0xc0170, 2016-08-08:
Please set properly the name of the issue.

> It isn't using my corrrect IAR path (my fault for not setting it in mbed_settings). I just think there should be a better error than this

+1

If we can check if paths are correctly set - check if a compiler returns something valid for instance for armcc it is a product version with some other information.

bridadan, 2016-08-08:
The funny thing is we actually do check to see if the path exists for the environment variables (ex `MBED_IAR_PATH`), just not for `mbed_settings.py`: https://github.com/ARMmbed/mbed-os/blob/master/tools/settings.py#L80

Seems like this wouldn't be that too hard of a change!

PR #2418 (2016-08-26): Readable error when toolchain paths not set. Allow IAR and ARM toolchains to be set in user's PATH.
Fixes #2360.

New error:
[ERROR] Could not find executable for IAR.
Currently set search path: /default/path/that/doesnt/exist
(System exit before any build system calls)

@theotherjimmy  @0xc0170 @bridadan
commit 14458868 (2016-08-26): Readable error when toolchain paths not set.

Fixes #2360.

New error:
[Error] Toolchain path does not exist for IAR.
Current value: /default/path/that/doesnt/exist
(System exit before any build system calls)
commit fb8194a3 (2016-08-30): Readable error when toolchain paths not set.

Fixes #2360.

New error:
[Error] Toolchain path does not exist for IAR.
Current value: /default/path/that/doesnt/exist
(System exit before any build system calls)
commit e78e0235 (2016-08-31): Readable error when toolchain paths not set.

Fixes #2360.

New error:
[Error] Toolchain path does not exist for IAR.
Current value: /default/path/that/doesnt/exist
(System exit before any build system calls)
closed 2016-08-26 by sg-