ARMmbed/mbed-os #2369: DNS failure when trying to get ip of local host

adamelhakham, 2016-08-04:
When running the following code:

	EthernetInterface ifaceHandler;
	char address[] = "prov-dev.kfn.arm.com";//" this is a local host, running in ARM network with IP 10.45.51.47"
	uint16_t port = 5683;
	SocketAddress *addr;
	const char *ipGet;

	ifaceHandler.connect();

	addr = new SocketAddress(&ifaceHandler, address, port);
	
	ipGet = addr->get_ip_address();
	
	printf("ip is: %s\n", ipGet);
	
the message "ip is: 0.0.0.0" is printed instead of "ip is: 10.45.51.47".
I tried digging a bit deeper and found that in the following function (in DnsQuery.cpp):
static int32_t query(UDPSocket *socket, const SocketAddress &addr, const char *hostname, char *ipaddress)

it receives an answer from the DNS but failes to resolve the packet and runs the following code:

if (!resolve(packet, ipaddress)) {
            delete packet;
            return NSAPI_ERROR_DNS_FAILURE;
}
	

* note that on public hosts everything works correctly.

@jenia81

bridadan, 2016-08-04:
cc @sarahmarshy @geky

geky, 2016-08-04:
Hi @adamelhakham, thanks for reporting this. This would be caused by not using DNS servers supplied by DHCP acknowledgement. This required a bit of tweaking internal to the network-socket API. I should be able to have a fix up soon.

sg-, 2016-08-26:
Should be fixed with #2497

closed 2016-08-26 by sg-