ARMmbed/mbed-os #2370: exporter listing improvements

BlackstoneEngineering, 2016-08-04:
## Problem
It is very hard to figure out what exporters are supported by a single board in the mbed-cli toolset.

#### Use case
I have a board plugged in, example code imported, and I want to export to a IDE to debug. I know the export command exists, but how do I know what exporters I can use? Currently I have to type `mbed export -i UVISION4 --supported` to get a list of all boards that uvision supports, but I dont want all boads, I just want to know about one board.

## Secondary Problem
The `mbed export -i ..... --supported` options overflows the standard windows window, even when I expand it to 2x its width, 160 characters, the lines still overflow, further the terminal is immediately line wrapped rendering the text garbled and meaningless, and worse yet I cant even see the top, i lose data and gain no knowledge from this other than frustration. 

## Solution
Here are some proposed solutions
1. enable `mbed export -m MCU --supported`, this would print a single line out of the matrix denoting what exporters the mcu supports
    * extend this such that `mbed export --supported` takes the implicit target currently set 
2. clean up output of current tool so that it doesnt overflow or line wrap and is ledgible, current method is unusable
3. support tab completion, so i could `mbed export -i <tab>` and the list of supported IDE's for the current MCU would cycle, or even `mbed export -m MCU -i <tab>` for a similar effect. Adding tab completion negates the need for the massive table and enables the user to interactively explore the tool's options. 

Point is I cant easily figure out what exporters, need to fix this. 

@sg- @screamerbg

geky, 2016-08-04:
cc @theotherjimmy who I believe has looked into this before

theotherjimmy, 2016-08-04:
> 1. enable `mbed export -m MCU --supported`

That's completely reasonable

> - extend this such that `mbed export --supported` takes the implicit target currently set.

There is also an implicit toolchain and possibly an implicit IDE. This makes it ambiguous which one should implicitly be picked up.

> 2. Clean up the output so it does not wrap

Yup, also completely possible

> 3. Tab completion

Good luck.

theotherjimmy, 2016-08-04:
We could also extend some of the error messages to tell the user to run `mbed export -m <foo> --supported` or `mbed export -i <bar> --supported` when we run into an unsupported target, ide combination.

theotherjimmy, 2016-08-04:
Both mbed-cli and the mbed tools need to be messed with to make this happen. https://github.com/ARMmbed/mbed-cli/blob/master/mbed/mbed.py#L2126 needs to take the `-m` and the `-i` into account, and https://github.com/ARMmbed/mbed-os/blob/master/tools/project.py#L103 also needs to take the `-m` and the `-i` into account.

closed 2017-10-27 by ChiefBureaucraticOfficer