ARMmbed/mbed-os #2374: Issue when using compile after building from IAR Workbench

ConorPKeegan, 2016-08-04:
When trying to compile an example using `mbed compile -m NUCLEO_F746ZG -t ARM`, errors occur due to the files in the `projectfiles/iar_NUCLEO_F746ZG` also being linked. This causes the compilation to fail.

Copy of ARMmbed/mbed-cli#317

# To reproduce error
* Have the project exported to IAR using `mbed export -m NUCLEO_F746ZG -i IAR`
* Build the project inside of IAR Workbench
* Try to compile the project again using the CLI with `mbed compile -m NUCLEO_F746ZG -t ARM`
* This causes the compilation to fail, but it will work if you delete the `projectfiles/` directory first

# Error Example
An example of the error that is produced is:
```
Warning: L6170W: Mapping symbol #156 '$t' in .bss(BLE.o:85) identifies code, but is in a section not marked as executable.
Error: L6654E: Rejected Local symbol .text9 referred to from non group member BLE.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text11 referred to from non group member BLE.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text65 referred to from non group member BLE.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text67 referred to from non group member BLE.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text69 referred to from non group member BLE.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text71 referred to from non group member BLE.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text73 referred to from non group member BLE.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text75 referred to from non group member BLE.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text77 referred to from non group member BLE.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text79 referred to from non group member BLE.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text9 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text11 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text13 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text15 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text17 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text19 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text21 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text23 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text25 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text27 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text29 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text31 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text33 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text35 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text37 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text39 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text41 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text43 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text45 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text47 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text49 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text51 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text53 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text55 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text57 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text59 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text61 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text63 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text65 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text67 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text69 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text71 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text73 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text118 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text120 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text122 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text124 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text126 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text128 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text130 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text132 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .debug_info222 referred to from non group member DiscoveredCharacteristic.o(.iar.debug_line)
Error: L6654E: Rejected Local symbol .debug_info227 referred to from non group member DiscoveredCharacteristic.o(.iar.debug_line)
Error: L6654E: Rejected Local symbol .debug_info242 referred to from non group member DiscoveredCharacteristic.o(.iar.debug_line)
Error: L6654E: Rejected Local symbol .debug_info242 referred to from non group member DiscoveredCharacteristic.o(.iar.debug_line)
Error: L6654E: Rejected Local symbol .debug_info247 referred to from non group member DiscoveredCharacteristic.o(.iar.debug_line)
[... 43 lines cut ...]
Error: L6654E: Rejected Local symbol .text63 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text65 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text67 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text69 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text71 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text73 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text118 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
Error: L6654E: Rejected Local symbol .text120 referred to from non group member DiscoveredCharacteristic.o(.iar.stackusage)
[... cut ...]
Error: L6654E: Rejected Local symbol .debug_info247 referred to from non group member DiscoveredCharacteristic.o(.iar.debug_line)
Error: L6654E: Rejected Local symbol .debug_info247 referred to from non group member DiscoveredCharacteristic.o(.iar.debug_line)
Error: L6654E: Rejected Local symbol .text10 referred to from non group member GapScanningParams.o(.iar.stackusage)
Warning: L6170W: Mapping symbol #136 '$t' in .bss(RTX_Conf_CM.o:48) identifies code, but is in a section not marked as executable.
Not enough information to list the image map.
Finished: 1 information, 2 warning and 58 error messages.
```

0xc0170, 2016-08-08:
I assume projectfiles folder is not excluded by exporters. We shall have a look and fix this, might be valid for all tools

sarahmarshy, 2016-08-11:
I was not able to recreate this issue. However, you should be able to place a ```.mbedignore``` file with the contents ```*``` in the directory you wish to be excluded from compilation. 

@0xc0170  Do you think we should by default put ```.mbedignore``` in the export directory?

sarahmarshy, 2016-08-15:
@ConorPKeegan can you verify if this fixes your issue and close this?

ConorPKeegan, 2016-08-19:
@sarahmarshy that fixed the problem for me, thanks.

closed 2016-08-19 by ConorPKeegan

0xc0170, 2016-08-19:
@ConorPKeegan However, if you export a project, and the project files are included by default, is a bug that needs fixing. adding ``.mbedignore`` should be done by exporters by default then.  I haven't seen this bug yet with exporters

ConorPKeegan, 2016-08-19:
@0xc0170 It is odd that this only happens to me, I have tried it on multiple examples with the same results.