ARMmbed/mbed-os #2383: All ble targets _BOOT and _OTA fail on greentea

jacobrosenthal, 2016-08-07:
```Compile: test_env.cpp
[Error] test_env.cpp@64,1: 'RawSerial' does not name a type
[Error] test_env.cpp@196,5: 'greentea_serial' was not declared in this scope
[Error] test_env.cpp@214,5: 'greentea_serial' was not declared in this scope
[Error] test_env.cpp@232,9: 'greentea_serial' was not declared in this scope
[Error] test_env.cpp@258,9: 'greentea_serial' was not declared in this scope
[Error] test_env.cpp@278,9: 'greentea_serial' was not declared in this scope
[Error] test_env.cpp@301,9: 'greentea_serial' was not declared in this scope
[Error] test_env.cpp@325,9: 'greentea_serial' was not declared in this scope
[Error] test_env.cpp@358,9: 'greentea_serial' was not declared in this scope
[Error] test_env.cpp@391,9: 'greentea_serial' was not declared in this scope
[ERROR] ./mbed-os/features/frameworks/greentea-client/source/test_env.cpp:64:1: error: 'RawSerial' does not name a type
 RawSerial greentea_serial(USBTX, USBRX); 
 ^
./mbed-os/features/frameworks/greentea-client/source/test_env.cpp: In function 'void greentea_write_preamble()':
./mbed-os/features/frameworks/greentea-client/source/test_env.cpp:196:5: error: 'greentea_serial' was not declared in this scope
     greentea_serial.putc('{');
     ^
./mbed-os/features/frameworks/greentea-client/source/test_env.cpp: In function 'void greentea_write_postamble()':
./mbed-os/features/frameworks/greentea-client/source/test_env.cpp:214:5: error: 'greentea_serial' was not declared in this scope
     greentea_serial.putc('}');
     ^
./mbed-os/features/frameworks/greentea-client/source/test_env.cpp: In function 'void greentea_write_string(const char*)':
./mbed-os/features/frameworks/greentea-client/source/test_env.cpp:232:9: error: 'greentea_serial' was not declared in this scope
         greentea_serial.putc(*str);
         ^
./mbed-os/features/frameworks/greentea-client/source/test_env.cpp: In function 'void greentea_write_int(int)':
./mbed-os/features/frameworks/greentea-client/source/test_env.cpp:258:9: error: 'greentea_serial' was not declared in this scope
         greentea_serial.putc(intval[i]);
         ^
./mbed-os/features/frameworks/greentea-client/source/test_env.cpp: In function 'void greentea_send_kv(const char*, const char*)':
./mbed-os/features/frameworks/greentea-client/source/test_env.cpp:278:9: error: 'greentea_serial' was not declared in this scope
         greentea_serial.putc(';');
         ^
./mbed-os/features/frameworks/greentea-client/source/test_env.cpp: In function 'void greentea_send_kv(const char*, int)':
./mbed-os/features/frameworks/greentea-client/source/test_env.cpp:301:9: error: 'greentea_serial' was not declared in this scope
         greentea_serial.putc(';');
         ^
./mbed-os/features/frameworks/greentea-client/source/test_env.cpp: In function 'void greentea_send_kv(const char*, const char*, int)':
./mbed-os/features/frameworks/greentea-client/source/test_env.cpp:325:9: error: 'greentea_serial' was not declared in this scope
         greentea_serial.putc(';');
         ^
./mbed-os/features/frameworks/greentea-client/source/test_env.cpp: In function 'void greentea_send_kv(const char*, const char*, int, int)':
./mbed-os/features/frameworks/greentea-client/source/test_env.cpp:358:9: error: 'greentea_serial' was not declared in this scope
         greentea_serial.putc(';');
         ^
./mbed-os/features/frameworks/greentea-client/source/test_env.cpp: In function 'void greentea_send_kv(const char*, int, int)':
./mbed-os/features/frameworks/greentea-client/source/test_env.cpp:391:9: error: 'greentea_serial' was not declared in this scope
         greentea_serial.putc(';');
         ^

[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u /Users/jacobrosenthal/mbed-os-example-ble/BLE_LED/mbed-os/tools/make.py -t GCC_ARM -m SEEED_TINY_BLE_BOOT --source . --build ./.build/SEEED_TINY_BLE_BOOT/GCC_ARM" in "/Users/jacobrosenthal/mbed-os-example-ble/BLE_LED"
---
Jacobs-MacBook-Air-3:BLE_LED jacobrosenthal$ mbed --version
0.9.1
```

jacobrosenthal, 2016-08-07:
Seems like maybe those arent supposed to be directly usable targets? Can I just add them as a second inherited target to the main target then? Then would one create a new target inheriting both without editing targets.json?

I had them offered up on the command line as seemingly valid options by:
```
Jacobs-MacBook-Air-3:BLE_LED jacobrosenthal$ mbed target DOGS
[mbed] DOGS now set as default target in program "BLE_LED"
Jacobs-MacBook-Air-3:BLE_LED jacobrosenthal$ mbed compile -t GCC_ARM
usage: make.py [-h] [-m MCU] [-t TOOLCHAIN] [--color] [-c] [-o OPTIONS]
               [-p PROGRAM] [-n PROGRAM] [-j JOBS] [-v] [--silent] [-D MACROS]
               [-S] [-f GENERAL_FILTER_REGEX] [--automated] [--host HOST_TEST]
               [--extra EXTRA] [--peripherals PERIPHERALS]
               [--dep DEPENDENCIES] [--source SOURCE_DIR]
               [--duration DURATION] [--build BUILD_DIR] [-N ARTIFACT_NAME]
               [-d DISK] [-s SERIAL] [-b BAUD] [-L] [--rtos] [--rpc] [--eth]
               [--usb_host] [--usb] [--dsp] [--fat] [--ublox] [--testlib]
               [-l LINKER_SCRIPT]
make.py: error: argument -m/--mcu: DOGS is not a supported MCU. Supported MCUs are:
ARCH_BLE,              ARCH_BLE_BOOT,         ARCH_BLE_OTA, 
ARCH_GPRS,             ARCH_LINK,             ARCH_LINK_BOOT, 
ARCH_LINK_OTA,         ARCH_MAX,              ARCH_PRO, 
ARM_BEETLE_SOC,        ARM_IOTSS_BEID,        ARM_MPS2_M0, 
ARM_MPS2_M0P,          ARM_MPS2_M1,           ARM_MPS2_M3, 
ARM_MPS2_M4,           ARM_MPS2_M7,           B96B_F446VE, 
BLUEPILL_F103C8,       CHIPPD_TINY_BLE,       CHIPPD_TINY_BLE_BOOT, 
CHIPPD_TINY_BLE_OTA,   DELTA_DFCM_NNN40,      DELTA_DFCM_NNN40_BOOT, 
DELTA_DFCM_NNN40_OTA,  DISCO_F051R8,          DISCO_F100RB, 
DISCO_F303VC,          DISCO_F334C8,          DISCO_F401VC, 
DISCO_F407VG,          DISCO_F429ZI,          DISCO_F469NI, 
DISCO_F746NG,          DISCO_L053C8,          DISCO_L476VG, 
EFM32GG_STK3700,       EFM32HG_STK3400,       EFM32LG_STK3600, 
EFM32PG_STK3401,       EFM32WG_STK3800,       EFM32ZG_STK3200, 
ELEKTOR_COCORICO,      ELMO_F411RE,           HEXIWEAR, 
HRM1017,               HRM1017_BOOT,          HRM1017_OTA, 
K20D50M,               K22F,                  K64F, 
KL05Z,                 KL25Z,                 KL26Z, 
KL27Z,                 KL43Z,                 KL46Z, 
LPC1114,               LPC11C24,              LPC11U24, 
LPC11U24_301,          LPC11U34_421,          LPC11U35_401, 
LPC11U35_501,          LPC11U35_501_IBDAP,    LPC11U35_Y5_MBUG, 
LPC11U37H_401,         LPC11U37_501,          LPC11U68, 
LPC1347,               LPC1549,               LPC1768, 
LPC2368,               LPC2460,               LPC4088, 
LPC4088_DM,            LPC4330_M0,            LPC4330_M4, 
LPC4337,               LPC810,                LPC812, 
LPC824,                LPCCAPPUCCINO,         MAX32600MBED, 
MAXWSNENV,             MICRONFCBOARD,         MOTE_L152RC, 
MTS_DRAGONFLY_F411RE,  MTS_GAMBIT,            MTS_MDOT_F405RG, 
MTS_MDOT_F411RE,       NRF51822,              NRF51822_BOOT, 
NRF51822_OTA,          NRF51822_Y5_MBUG,      NRF51_DK, 
NRF51_DK_BOOT,         NRF51_DK_LEGACY,       NRF51_DK_OTA, 
NRF51_DONGLE,          NRF51_DONGLE_BOOT,     NRF51_DONGLE_OTA, 
NRF51_MICROBIT,        NRF51_MICROBIT_B,      NRF51_MICROBIT_BOOT, 
NRF51_MICROBIT_B_BOOT, NRF51_MICROBIT_B_OTA,  NRF51_MICROBIT_OTA, 
NRF52_DK,              NUCLEO_F030R8,         NUCLEO_F031K6, 
NUCLEO_F042K6,         NUCLEO_F070RB,         NUCLEO_F072RB, 
NUCLEO_F091RC,         NUCLEO_F103RB,         NUCLEO_F207ZG, 
NUCLEO_F302R8,         NUCLEO_F303K8,         NUCLEO_F303RE, 
NUCLEO_F334R8,         NUCLEO_F401RE,         NUCLEO_F410RB, 
NUCLEO_F411RE,         NUCLEO_F429ZI,         NUCLEO_F446RE, 
NUCLEO_F446ZE,         NUCLEO_F746ZG,         NUCLEO_F767ZI, 
NUCLEO_L011K4,         NUCLEO_L031K6,         NUCLEO_L053R8, 
NUCLEO_L073RZ,         NUCLEO_L152RE,         NUCLEO_L432KC, 
NUCLEO_L476RG,         NUMAKER_PFM_NUC472,    NZ32_SC151, 
OC_MBUINO,             RBLAB_BLENANO,         RBLAB_BLENANO_BOOT, 
RBLAB_BLENANO_OTA,     RBLAB_NRF51822,        RBLAB_NRF51822_BOOT, 
RBLAB_NRF51822_OTA,    RZ_A1H,                SAMD21G18A, 
SAMD21J18A,            SAMG55J19,             SAML21J18A, 
SAMR21G18A,            SEEED_TINY_BLE,        SEEED_TINY_BLE_BOOT, 
SEEED_TINY_BLE_OTA,    SSCI824,               STM32F3XX, 
STM32F407,             TEENSY3_1,             TY51822R3, 
TY51822R3_BOOT,        TY51822R3_OTA,         UBLOX_C027, 
UBLOX_C029,            VK_RZ_A1H,             WALLBOT_BLE, 
WALLBOT_BLE_BOOT,      WALLBOT_BLE_OTA,       WIZWIKI_W7500, 
WIZWIKI_W7500ECO,      WIZWIKI_W7500P,        XADOW_M0, 
XBED_LPC1768           
[mbed] ERROR: "python" returned error code 2.
[mbed] ERROR: Command "python -u /Users/jacobrosenthal/mbed-os-example-ble/BLE_LED/mbed-os/tools/make.py -t GCC_ARM -m DOGS --source . --build ./.build/DOGS/GCC_ARM" in "/Users/jacobrosenthal/mbed-os-example-ble/BLE_LED"
---
Jacobs-MacBook-Air-3:BLE_LED jacobrosenthal$ 
```

pan-, 2016-08-08:
NRF51 targets with only 16K of RAM are not supported on mbed OS 5 but still are on mbed OS 2.
Greentea is a a feature of mbed OS 5 not 2. 

Nevertheless, it is strange to say the less that you have an error  in this area. 
After a quick look in `hal/targets.json`, it appears that the device target capabilities are correctly defined for the `SEEED_TINY_BLE` [target](https://github.com/ARMmbed/mbed-os/blob/master/hal/targets.json#L1349) but no capabilities are defined by the `_BOOT` and `_OTA` targets.

bridadan, 2016-08-08:
@jacobrosenthal Thanks for reporting this!

@pan- I noticed this as well, I'm guessing this just got removed at some point. Sounds like it'd be a simple fix!

sg-, 2016-09-19:
I think this needs target.json input validation to reject commands that are not valid.

closed 2017-10-27 by ChiefBureaucraticOfficer