ARMmbed/mbed-os #2385: mbed-os-example-uvisor produces broken .bin file against 5.1, but not 5.0-rc5

janjongboom, 2016-08-08:
From https://forums.mbed.com/t/mbed-os-example-uvisor-problem-with-mbed-os-5-1-0/1758

I'm on an OS X box, and cannot build a working mbed-os-example-uvisor if I use the latest mbed os which is currently 5.1.0. I'm using a FRDM-K64F board.

This results is a broken .bin which won't run:

mbed import https://github.com/ARMmbed/mbed-os-example-uvisor
cd mbed-os-example-uvisor
mbed compile -m K64F -t GCC_ARM

However if I downgrade to 5.1.0-rc5 then I get a working image:

mbed update mbed-os-5.1.0-rc5
mbed compile -m K64F -t GCC_ARM

I could give more information if it helps. I just installed all of the development tools this week, and was able to create a working build yesterday. I was surprised when it broke today.

0xc0170, 2016-08-08:
cc @AlessandroA @meriac

AlessandroA, 2016-08-08:
Commit 88564a9ac37d4e65ce13b01eda7cdf7270a33129 seems to be introducing the regression (found with `git bisect`).

If you run `mbed compile -m K64F -t GCC_ARM -vv` and check the output for the startup ASM file:

```text
[DEBUG] Compile: arm-none-eabi-gcc -x assembler-with-cpp -c -Wall -Wextra
-Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0
-fno-exceptions -fno-builtin -ffunction-sections -fdata-sections -funsigned-char
-MMD -fno-delete-null-pointer-checks -fomit-frame-pointer -mcpu=cortex-m4
-mthumb -g -O0 -D__CORTEX_M4 -DCPU_MK64FN1M0VMD12 -DARM_MATH_CM4
-D__MBED_CMSIS_RTOS_CM -D__CMSIS_RTOS -DFSL_RTOS_MBED
-DMBEDTLS_ENTROPY_HARDWARE_ALT 
@./.build/K64F/GCC_ARM/.includes_0841d7d9675c9ac2e2f7a22f2192d0b6.txt -include ./.build/K64F/GCC_ARM/mbed_config.h
-o ./.build/K64F/GCC_ARM/mbed-os/hal/targets/cmsis/TARGET_Freescale/TARGET_K64F/TOOLCHAIN_GCC_ARM/startup_MK64F12.o
./mbed-os/hal/targets/cmsis/TARGET_Freescale/TARGET_K64F/TOOLCHAIN_GCC_ARM/startup_MK64F12.S
```

it looks like the command line does not define explicitly `FETURE_UVISOR` and `TARGET_UVISOR_SUPPORTED`. These 2 symbols should be triggered by the config file `mbed_app.json`.

If you look at the `mbed_config.h` file, which is force-included, those symbols are not defined either.

In the previous working version, `mbed-os-5.1.0-rc5`, the symbols were defined on the command line:

```text
[DEBUG] Compile: arm-none-eabi-gcc -x assembler-with-cpp -c -Wall -Wextra
-Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0
-fno-exceptions -fno-builtin -ffunction-sections -fdata-sections -funsigned-char
-MMD -fno-delete-null-pointer-checks -fomit-frame-pointer -mcpu=cortex-m4
-mthumb -g -O0 -DTARGET_KSDK2_MCUS -D__MBED__=1 -DDEVICE_I2CSLAVE=1
-DTARGET_LIKE_MBED -DTARGET_Freescale -DDEVICE_PORTINOUT=1
-DTARGET_RTOS_M4_M7 -DDEVICE_RTC=1 -DTOOLCHAIN_object -D__CMSIS_RTOS
-DFSL_RTOS_MBED -DDEVICE_STORAGE=1 -DMBEDTLS_ENTROPY_HARDWARE_ALT
-DMBED_BUILD_TIMESTAMP=1470659142.92 -DTARGET_KPSDK_MCUS
-DTOOLCHAIN_GCC -DTARGET_CORTEX_M -DTARGET_DEBUG
-DTARGET_LIKE_CORTEX_M4 -DDEVICE_ANALOGOUT=1 -DTARGET_M4
-DTARGET_K64F -DDEVICE_SERIAL=1 -DDEVICE_INTERRUPTIN=1 -DDEVICE_I2C=1
-DDEVICE_PORTOUT=1 -D__CORTEX_M4 -DDEVICE_STDIO_MESSAGES=1
-DCPU_MK64FN1M0VMD12 -DFEATURE_IPV4=1 -DTARGET_UVISOR_SUPPORTED
-DTARGET_FF_ARDUINO -DTARGET_KPSDK_CODE -DDEVICE_SERIAL_FC=1
-DFEATURE_STORAGE=1 -D__MBED_CMSIS_RTOS_CM -DDEVICE_SLEEP=1
-DTOOLCHAIN_GCC_ARM -DTARGET_FRDM -DDEVICE_SPI=1 -DFEATURE_UVISOR=1
-DDEVICE_ERROR_RED=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1
-DDEVICE_PWMOUT=1 -DDEVICE_PORTIN=1 -DTARGET_MCU_K64F -DARM_MATH_CM4 
@./.build/K64F/GCC_ARM/.includes_0841d7d9675c9ac2e2f7a22f2192d0b6.txt
-include ./.build/K64F/GCC_ARM/mbed_config.h -o ./.build/K64F/GCC_ARM/mbed-os/hal/targets/cmsis/TARGET_Freescale/TARGET_K64F/TOOLCHAIN_GCC_ARM/startup_MK64F12.o
./mbed-os/hal/targets/cmsis/TARGET_Freescale/TARGET_K64F/TOOLCHAIN_GCC_ARM/startup_MK64F12.S
```

I guess this is related to #2347.

@0xc0170 @sg-

sg-, 2016-08-08:
@janjongboom @AlessandroA for now adding ```FEATURE_UVISOR``` and ```TARGET_UVISOR_SUPPORTED``` in a macros_add you should be able to continue. We need to update docs and better define what emits a -D and what is used for directory scanning in the build system. These 2 should not be treated the same as they currently are.

sg-, 2016-08-26:
should be resolved with latest mbed-os-example-uvisor

closed 2016-08-26 by sg-