ARMmbed/mbed-os #2388: Compile error on latest mbed-os with FEATURE_CLIENT enabled

LiyouZhou, 2016-08-08:
Steps to reproduce:

```
mbed new myprog
cd myprog
edit mbed_app.json with the following content
{
    "target_overrides": {
        "*": {
            "target.features_add": ["IPV6", "CLIENT", "IPV4"]
        }
    }
}
mbed compile -t GCC_ARM -m K64F
```
Error
```
Building project myprog (K64F, GCC_ARM)
Scan: .
Scan: FEATURE_BLE
Scan: FEATURE_UVISOR
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_CLIENT
Scan: FEATURE_STORAGE
Scan: FEATURE_IPV4
Scan: FEATURE_IPV6
Scan: mbed
Scan: env
Compile: m2mconnectionhandler.cpp
[ERROR] In file included from ./mbed-os/features/FEATURE_CLIENT/mbed-client-classic/source/m2mconnectionhandler.cpp:16:0:
./mbed-os/features/FEATURE_CLIENT/mbed-client-classic/mbed-client-classic/m2mconnectionhandlerpimpl.h:19:22: fatal error: ns_types.h: No such file or directory
compilation terminated.

[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u /tmp/myprog/mbed-os/tools/make.py -t GCC_ARM -m K64F --source . --build ./.build/K64F/GCC_ARM" in "/tmp/myprog"
---
```

mbed-os version: `https://github.com/ARMmbed/mbed-os/#0712b8adf6bbc7eb796d5dac26f95d79d40745ef`
mbed-cli version: 0.9.1

0xc0170, 2016-08-09:
cc @yogpan01

LiyouZhou, 2016-08-09:
It seems I need to add "FEATURE_COMMON_PAL" in the target features list. `Scan: FEATURE_COMMON_PAL` was printed but it did not actually get scanned.

jupe, 2016-08-13:
yes `FEATURE_COMMON_PAL` feature is needed because mbed-trace and mbed-client-c etc is moved behind it.

sg-, 2016-08-26:
Please use ```"target.features_add": ["IPV6", "CLIENT", "IPV4", "COMMON_PAL"]```

closed 2016-08-26 by sg-