ARMmbed/mbed-os #2390: Add Support For Discovery F411 with the  STM32F411VET6

thelme, 2016-08-08:
Could you add mbed Support For Discovery F411 board ?

0xc0170, 2016-08-09:
cc @bcostm @adustm @LMESTM

bcostm, 2016-09-09:
Hi,
This board is not planned to be supported on mbed for now. We'll put availability date directly on the board page:
https://developer.mbed.org/platforms/ST-Discovery-F411E/
Can you please close this issue ?

thelme, 2016-09-09:
Ok, thanck you.

closed 2016-09-09 by thelme

jlunder, 2017-06-21:
Hi -- is there a reason this board is not supported? I have a couple and I can't imagine there's much to the port, given the cross compatibility between STM32 MCUs. Can I assist with a PR?

bcostm, 2017-06-21:
Hi,
I confirm it is not in ST plan to publish this board on mbed for the moment and it is not a technical issue. There is already the NUCLEO_F411RE board and the porting should be easy. But for sure you are more than welcome to help :). You can send a PR and we'll review it. Thanks.