ARMmbed/mbed-os #2396: Please update mbed-os to have stdio baud rate configurable

markus-becker-tridonic-com, 2016-08-09:
Moved from https://github.com/ARMmbed/mbed-os-example-blinky/issues/23

markus-becker-tridonic-com, 2016-08-09:
Small a example as requested:

main.cpp:
```
#include "mbed.h"

#define HAVE_DEBUG 1
#include "mbed-trace/mbed_trace.h"
//#include "mbed-client-libservice/ns_trace.h"
#define TRACE_GROUP "stdio"

Serial output(USBTX, USBRX);

int main() {
    //output.baud(115200); // with or without
    output.printf("output.printf()\n");
    printf("printf\n");

    mbed_trace_init();
    tr_debug("tr_debug\n");

    output.printf("output.printf()2\n");
    printf("printf2\n");
}
```

mbed_app.json
```
{
    "config": {
        "core": {
            "stdio-baud-rate": 115200 // with or without
        }
    },
    "target_overrides": {
        "*": {
            "target.features_add": ["COMMON_PAL"],
            "mbed-trace.enable": 1
        }
    }
}
```

markus-becker-tridonic-com, 2016-08-09:
Results in:

BAD:
* Without output.baudrate(115200)
* With config core stdio-baud-rate 115200
* Terminal configured for 9600
output.printf()<\n>
printf<\n>
<\r><27>[2K<27>[90m[DBG ][stdio]: tr_debug<\n>
<27>[0m<\n>
output.printf()2<\n>
printf2<\n>
--> Should be terminal configured for 115200.

GOOD:
* With output.baudrate(115200)
* With config core stdio-baud-rate 115200
* Terminal configured for 115200
output.printf()<\n>
printf<\n>
<\r><27>[2K<27>[90m[DBG ][stdio]: tr_debug<\n>
<27>[0m<\n>
output.printf()2<\n>
printf2<\n>

GOOD:
* With output.baudrate(115200)
* Without config core stdio-baud-rate 115200
* Terminal configured for 115200
output.printf()<\n>
printf<\n>
<\r><27>[2K<27>[90m[DBG ][stdio]: tr_debug<\n>
<27>[0m<\n>
output.printf()2<\n>
printf2<\n>

GOOD:
* Without output.baudrate(115200)
* Without config core stdio-baud-rate 115200
* Terminal configured for 9600
output.printf()<\n>
printf<\n>
<\r><27>[2K<27>[90m[DBG ][stdio]: tr_debug<\n>
<27>[0m<\n>
output.printf()2<\n>
printf2<\n>

markus-becker-tridonic-com, 2016-08-09:
Additionally when debugging mbed-mesh-api you get mixed 9600 and 115200 output.

markus-becker-tridonic-com, 2016-08-09:
@marcelosalazar

sg-, 2016-08-26:
Should be resolved.

closed 2016-08-26 by sg-