ARMmbed/mbed-os #2399: How to retrieve the remote host's IP for a TCP connection on the server side?

tommikas, 2016-08-09:
I.e. after using a TCPServer instance to accept a new TCP connection, how can I retrieve the IP of the remote host? Looking at Socket, TCPServer and TCPSocket classes there doesn't seem to be an obvious mechanism for it.

0xc0170, 2016-08-09:
cc @geky

geky, 2016-08-09:
This is a good question, and by good question, I mean this is a **good** question.

Most uses of the TCP server generally don't care where the incoming connection is from, so no one must have ran into this requirement yet. Thanks for raising the issue, as this functionality should be available.

Fortunately, this will be an easy API change to make, through overloading the TCPServer accept method:
``` cpp
/** Accepts a connection on a TCP socket
     *
     *  The server socket must be bound and set to listen for connections.
     *  On a new connection, creates a network socket using the specified
     *  socket instance.
     *
     *  By default, accept blocks until data is sent. If socket is set to
     *  non-blocking or times out, NSAPI_ERROR_WOULD_BLOCK is returned
     *  immediately.
     *
     *  @param socket   TCPSocket instance that will handle the incoming connection.
     *  @param          Destination for the connection address or NULL
     *  @return         0 on success, negative error code on failure
     */
int accept(TCPSocket *connection, SocketAddress *address = NULL);
```

Let me know if you have any thoughts.

tommikas, 2016-08-10:
Cool :) That looks fine.

infinnovation, 2016-08-10:
I've just come across the same issue while trying to migrate applications from mbed classic (using the separate EthernetInterface library) to mbed os 5.

Adding support would also surely require a change to the underlying nsapi layer (and therefore the several stack implementations), since the socket_accept method does not pass any address information.

sg-, 2016-08-25:
@tommikas @infinnovation This should be resolved.

closed 2016-08-25 by sg-