ARMmbed/mbed-os #2403: . .

sarahmarshy, 2016-08-09:


closed 2016-08-09 by sarahmarshy

theotherjimmy, 2016-08-09:
...

bogdanm, 2016-08-09:
.. -. - . .-. . ... - .. -. --.

sarahmarshy, 2016-08-09:
PROTIP: ```cd ..```  creates a new issue titled '..'

sarahmarshy, 2016-08-09:
Github is cruel...

c1728p9, 2016-08-09:
-... --- --. -.. .- -. -- / -.. --- / -.-- --- ..- / .-. . .- .-.. .-.. -.-- / -.- -. --- .-- / -- --- .-. ... . / -.-. --- -.. . / --- .-. / -.. .. -.. / -.-- --- ..- / .--- ..- ... - / ..- ... . / .- / - .-. .- -. ... .-.. .- - --- .-.

bridadan, 2016-08-09:
Can you attach the verbose log for this? I'm having trouble reproducing locally...

sarahmarshy, 2016-08-09:
@bridadan I ran this sequence of commands on ubuntu 16.04:
```
rm -rf /
cd .. 
```

bridadan, 2016-08-09:
Ok, going to try that now. Will report back with what I find.

geky, 2016-08-09:
Ah, there's your issue

There has been a recent patch to the bash builtins, you will need to add --no-preserve-root like so:
```
rm -rf /
cd .. --no-preserve-root
```