ARMmbed/mbed-os #2406: "no target defined" error when building NUCLEO_F042K6 with GCC_ARM

mbains, 2016-08-10:
Unable to build basic blinky for NUCLEO_F042K6 with offline mbed-cli (v0.9.1)

Steps to reproduce:
1. `mbed import mbed-os-example-blinky`
2. `cd mbed-os-example-blink`
3. `mbed compile -t GCC_ARM -m NUCLEO_F042K6  -v`

```
(env) manny@linux-wh4c ~/c/m/mbed-os-example-blinky> mbed compile -t GCC_ARM -m NUCLEO_F042K6  -v
[mbed] Working path "/home/manny/code/mbed/mbed-os-example-blinky" (library)
[mbed] Exec "python -u /home/manny/code/mbed/mbed-os-example-blinky/mbed-os/tools/make.py -t GCC_ARM -m NUCLEO_F042K6 --source . --build ./.build/NUCLEO_F042K6/GCC_ARM -v" in /home/manny/code/mbed/mbed-os-example-blinky
Building project mbed-os-example-blinky (NUCLEO_F042K6, GCC_ARM)
Scan: .
Scan: FEATURE_BLE
Scan: FEATURE_CLIENT
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_UVISOR
Scan: FEATURE_IPV4
Scan: FEATURE_IPV6
Scan: FEATURE_STORAGE
Scan: mbed
Scan: env
[DEBUG] Macros: -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DDEVICE_RTC_LSI=1 -DDEVICE_PORTINOUT=1 -D__MBED_CMSIS_RTOS_CM -DTARGET_STM32F0 -DDEVICE_RTC=1 -DTOOLCHAIN_object -D__CMSIS_RTOS -DTOOLCHAIN_GCC -DDEVICE_CAN=1 -DTARGET_CORTEX_M -DARM_MATH_CM0 -DTARGET_UVISOR_UNSUPPORTED -DMBED_BUILD_TIMESTAMP=1470802127.46 -DTARGET_M0 -DDEVICE_SERIAL=1 -DDEVICE_INTERRUPTIN=1 -D__CORTEX_M0 -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -DTARGET_NUCLEO_F042K6 -DDEVICE_STDIO_MESSAGES=1 -DTARGET_FF_ARDUINO -DTARGET_RELEASE -DTARGET_STM -DDEVICE_PORTIN=1 -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DDEVICE_SPI=1 -DMBED_RTOS_SINGLE_THREAD -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DTARGET_LIKE_CORTEX_M0 -DTARGET_STM32F042K6
Compile: RTX_Conf_CM.c
[DEBUG] Compile: arm-none-eabi-gcc -std=gnu99 -c -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -fno-builtin -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks -fomit-frame-pointer -mcpu=cortex-m0 -mthumb -Os -DMBED_RTOS_SINGLE_THREAD -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DDEVICE_RTC_LSI=1 -DDEVICE_PORTINOUT=1 -D__MBED_CMSIS_RTOS_CM -DTARGET_STM32F0 -DDEVICE_RTC=1 -DTOOLCHAIN_object -D__CMSIS_RTOS -DTOOLCHAIN_GCC -DDEVICE_CAN=1 -DTARGET_CORTEX_M -DARM_MATH_CM0 -DTARGET_UVISOR_UNSUPPORTED -DMBED_BUILD_TIMESTAMP=1470802127.46 -DTARGET_M0 -DDEVICE_SERIAL=1 -DDEVICE_INTERRUPTIN=1 -D__CORTEX_M0 -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -DTARGET_NUCLEO_F042K6 -DDEVICE_STDIO_MESSAGES=1 -DTARGET_FF_ARDUINO -DTARGET_RELEASE -DTARGET_STM -DDEVICE_PORTIN=1 -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DDEVICE_SPI=1 -DMBED_RTOS_SINGLE_THREAD -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DTARGET_LIKE_CORTEX_M0 -DTARGET_STM32F042K6 @./.build/NUCLEO_F042K6/GCC_ARM/.includes_d09e2452754f60ca3b246f831f7c580d.txt -include ./.build/NUCLEO_F042K6/GCC_ARM/mbed_config.h -MD -MF ./.build/NUCLEO_F042K6/GCC_ARM/mbed-os/rtos/rtx/TARGET_CORTEX_M/RTX_Conf_CM.d -o ./.build/NUCLEO_F042K6/GCC_ARM/mbed-os/rtos/rtx/TARGET_CORTEX_M/RTX_Conf_CM.o ./mbed-os/rtos/rtx/TARGET_CORTEX_M/RTX_Conf_CM.c
[Error] RTX_Conf_CM.c@107,6: #error "no target defined"
[Error] RTX_Conf_CM.c@274,6: #error "no target defined"
[Error] RTX_CM_lib.h@122,32: 'OS_CLOCK' undeclared here (not in a function)
[Error] RTX_CM_lib.h@555,2: #error "no target defined"
[Error] RTX_CM_lib.h@582,48: 'OS_MAINSTKSIZE' undeclared (first use in this function)
[Error] RTX_CM_lib.h@588,48: 'INITIAL_SP' undeclared (first use in this function)
[Warning] RTX_CM_lib.h@713,5: implicit declaration of function 'atexit' [-Wimplicit-function-declaration]
[DEBUG] Return: 1
[DEBUG] Output: ./mbed-os/rtos/rtx/TARGET_CORTEX_M/RTX_Conf_CM.c:107:6: error: #error "no target defined"
[DEBUG] Output:  #    error "no target defined"
[DEBUG] Output:       ^
[DEBUG] Output: ./mbed-os/rtos/rtx/TARGET_CORTEX_M/RTX_Conf_CM.c:274:6: error: #error "no target defined"
[DEBUG] Output:  #    error "no target defined"
[DEBUG] Output:       ^
[DEBUG] Output: In file included from ./mbed-os/rtos/rtx/TARGET_CORTEX_M/RTX_Conf_CM.c:408:0:
[DEBUG] Output: ./mbed-os/rtos/rtx/TARGET_CORTEX_M/RTX_CM_lib.h:122:32: error: 'OS_CLOCK' undeclared here (not in a function)
[DEBUG] Output:  uint32_t const os_tickfreq   = OS_CLOCK;
[DEBUG] Output:                                 ^
[DEBUG] Output: ./mbed-os/rtos/rtx/TARGET_CORTEX_M/RTX_CM_lib.h:555:2: error: #error "no target defined"
[DEBUG] Output:  #error "no target defined"
[DEBUG] Output:   ^
[DEBUG] Output: ./mbed-os/rtos/rtx/TARGET_CORTEX_M/RTX_CM_lib.h: In function 'set_main_stack':
[DEBUG] Output: ./mbed-os/rtos/rtx/TARGET_CORTEX_M/RTX_CM_lib.h:582:48: error: 'OS_MAINSTKSIZE' undeclared (first use in this function)
[DEBUG] Output:      uint32_t interrupt_stack_size = ((uint32_t)OS_MAINSTKSIZE * 4);
[DEBUG] Output:                                                 ^
[DEBUG] Output: ./mbed-os/rtos/rtx/TARGET_CORTEX_M/RTX_CM_lib.h:582:48: note: each undeclared identifier is reported only once for each function it appears in
[DEBUG] Output: ./mbed-os/rtos/rtx/TARGET_CORTEX_M/RTX_CM_lib.h:588:48: error: 'INITIAL_SP' undeclared (first use in this function)
[DEBUG] Output:      uint32_t heap_plus_stack_size = ((uint32_t)INITIAL_SP - (uint32_t)HEAP_START) - interrupt_stack_size;
[DEBUG] Output:                                                 ^
[DEBUG] Output: ./mbed-os/rtos/rtx/TARGET_CORTEX_M/RTX_CM_lib.h: In function 'pre_main':
[DEBUG] Output: ./mbed-os/rtos/rtx/TARGET_CORTEX_M/RTX_CM_lib.h:713:5: warning: implicit declaration of function 'atexit' [-Wimplicit-function-declaration]
[DEBUG] Output:      atexit(__libc_fini_array);
[DEBUG] Output:      ^
Traceback (most recent call last):
  File "/home/manny/code/mbed/mbed-os-example-blinky/mbed-os/tools/make.py", line 271, in <module>
    name=options.artifact_name)
  File "/home/manny/code/mbed/mbed-os-example-blinky/mbed-os/tools/build_api.py", line 349, in build_project
    objects = toolchain.compile_sources(resources, build_path, resources.inc_dirs)
  File "/home/manny/code/mbed/mbed-os-example-blinky/mbed-os/tools/toolchains/__init__.py", line 756, in compile_sources
    return self.compile_seq(queue, objects)
  File "/home/manny/code/mbed/mbed-os-example-blinky/mbed-os/tools/toolchains/__init__.py", line 770, in compile_seq
    res['command']
  File "/home/manny/code/mbed/mbed-os-example-blinky/mbed-os/tools/toolchains/__init__.py", line 901, in compile_output
    raise ToolException(_stderr)
ToolException: ./mbed-os/rtos/rtx/TARGET_CORTEX_M/RTX_Conf_CM.c:107:6: error: #error "no target defined"
 #    error "no target defined"
      ^
./mbed-os/rtos/rtx/TARGET_CORTEX_M/RTX_Conf_CM.c:274:6: error: #error "no target defined"
 #    error "no target defined"
      ^
In file included from ./mbed-os/rtos/rtx/TARGET_CORTEX_M/RTX_Conf_CM.c:408:0:
./mbed-os/rtos/rtx/TARGET_CORTEX_M/RTX_CM_lib.h:122:32: error: 'OS_CLOCK' undeclared here (not in a function)
 uint32_t const os_tickfreq   = OS_CLOCK;
                                ^
./mbed-os/rtos/rtx/TARGET_CORTEX_M/RTX_CM_lib.h:555:2: error: #error "no target defined"
 #error "no target defined"
  ^
./mbed-os/rtos/rtx/TARGET_CORTEX_M/RTX_CM_lib.h: In function 'set_main_stack':
./mbed-os/rtos/rtx/TARGET_CORTEX_M/RTX_CM_lib.h:582:48: error: 'OS_MAINSTKSIZE' undeclared (first use in this function)
     uint32_t interrupt_stack_size = ((uint32_t)OS_MAINSTKSIZE * 4);
                                                ^
./mbed-os/rtos/rtx/TARGET_CORTEX_M/RTX_CM_lib.h:582:48: note: each undeclared identifier is reported only once for each function it appears in
./mbed-os/rtos/rtx/TARGET_CORTEX_M/RTX_CM_lib.h:588:48: error: 'INITIAL_SP' undeclared (first use in this function)
     uint32_t heap_plus_stack_size = ((uint32_t)INITIAL_SP - (uint32_t)HEAP_START) - interrupt_stack_size;
                                                ^
./mbed-os/rtos/rtx/TARGET_CORTEX_M/RTX_CM_lib.h: In function 'pre_main':
./mbed-os/rtos/rtx/TARGET_CORTEX_M/RTX_CM_lib.h:713:5: warning: implicit declaration of function 'atexit' [-Wimplicit-function-declaration]
     atexit(__libc_fini_array);
     ^

[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u /home/manny/code/mbed/mbed-os-example-blinky/mbed-os/tools/make.py -t GCC_ARM -m NUCLEO_F042K6 --source . --build ./.build/NUCLEO_F042K6/GCC_ARM -v" in "/home/manny/code/mbed/mbed-os-example-blinky"
---
```

**I'm using gcc version 5.2.1 20151202 and mbed-cli 0.9.1 installed with pip**

0xc0170, 2016-08-10:
Target ``NUCLEO_F042K6`` is not supported by rtos.

cc @bcostm @adustm @LMESTM

mbains, 2016-08-10:
@0xc0170 Thanks. I should have read the release notes :-) 
Any unofficial estimate of when it might get put in?

bcostm, 2016-09-09:
Hi,
mbed RTOS is not compatible with this platform (code size is too big for this device).

adustm, 2017-01-03:
Hello
It looks like this request won't happen and the issue won't be solved (target too small for the request).
Can someone close the issue, please ?
Cheers
Armelle

closed 2017-01-03 by bridadan