ARMmbed/mbed-os #2408: TCPServer.bind fails sometimes

kudangun, 2016-08-10:
I create a TCPServer and TCPSocket.
call server.open(), server.bind(port), server.listen(1) and server.accept().
Everything works OK. When i detect that the remote host closed its socket, i close() and delete my socket and then close() and delete the server. Then I start over - creating server and socket, open,bind etc.
Sometimes, after a few such iterations, server.bind(port) fails with status -3003.
This behavior is inconsistent. Sometimes it happens after 8 iterations, sometimes more. Sometimes not at all.
Using mbed os tag 0712b8adf6bbc7eb796d5dac26f95d79d40745ef
K64F, GCC_ARM

0xc0170, 2016-08-10:
cc @geky

geky, 2016-08-13:
Hi @kudangun, thanks for reporting this issue.

Network sockets have a mechanism to prevent new server instances from accidentally accepting sockets to old server instances. Think of if you start one program after another, the new program could accidentally get the old programs connections and corrupt the new program if the program wasn't written well.

This mechanism is time based, so if enough time has passed between the server has been taken down, a bind to the same address may succeed.

Once #2440 is merged, you can use `NSAPI_REUSEADDR` to allow socket servers to reuse the same address without waiting.

kudangun, 2016-08-15:
Hi @geky 
In this scenario, the remote host closes the socket and I respond by closing my socket. There are no old connections. I don't see why I shouldn't be able to create a new socket server and bind it to the same port. this is not address reuse

sg-, 2016-08-25:
@kudangun does #2440 solve this? Please file these issues with a test case or example program or code snippet to reproduce.

geky, 2016-08-29:
Unfortunately, this is just how the BSD sockets operate, in order to avoid problems with delayed TCP packets that could corrupt unrelated connections.

In BSD socket terminology, you are reusing a socket address (ip address + port) that a previous server ran on in the lifetime of the system, even if the previous server had been shut down.

If you want to respond to multiple connections, you should keep either keep the TCP server up for as long as you expect new connections, or if you really need TCPServer instances to be isolated, use the `NSAPI_REUSEADDR` flag.

Here's a more in-depth explanation of the issue TCP is trying to avoid:
http://stackoverflow.com/a/3233022

sg-, 2016-09-19:
@kudangun is this resolved?

closed 2017-01-16 by sg-