ARMmbed/mbed-os #2411: OS can break the compilation

javier-moreno-tridonic-com, 2016-08-10:
Trying to resolve this [issue](https://github.com/ARMmbed/mbed-os-example-client/issues/37) and thanks to @hasnainvirk we saw that when compiling some examples, the #include directives resolution has different behaviours in linux and windows. This is mainly caused by names that will be considered as different by linux and not by windows.

For instance the file thread.h that is part of the **sal-stack-nanostack-private** source and  Thread.h included in **mbed-os** source

MarceloSalazar, 2016-08-10:
@screamerbg @sg- can you look into this please?

sg-, 2016-08-12:
Thanks for the note. Need to update the includes client and rf driver use. Will resolve this @geky

geky, 2016-08-26:
The client has had conflicting header files removed for a while, but it unfortunately took a while to move into mainstream mbed-os.

I'm unfamiliar with the rf driver and it my have been removed in a previous change. Do the includes still present issues?

javier-moreno-tridonic-com, 2016-08-26:
I tried a fresh checkout of **mbed-os-example-client** and v5.0.5 of **sal-stack-nanostack-private**

Builds, but linking fails.

>Link: mbed-os
../.build/NUCLEO_F401RE/GCC_ARM/hal/common/retarget.o: In function `__wrap_main':
retarget.cpp:(.text.__wrap_main+0xe): undefined reference to `main'
collect2.exe: error: ld returned 1 exit status
[ERROR] ../.build/NUCLEO_F401RE/GCC_ARM/hal/common/retarget.o: In function `__wrap_main':
retarget.cpp:(.text.__wrap_main+0xe): undefined reference to `main'
collect2.exe: error: ld returned 1 exit status

>[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u C:\Code\sandbox\nanostacktest\mbed-os-example-client\mbed-os\tools\make.py -t GCC_ARM -m NUCLEO_F401RE --source .. --build ..\.build\NUCLEO_F401RE\GCC_ARM" in "C:\Code\sandbox\nanostacktest\mbed-os-example-client\mbed-os\features"

hasnainvirk, 2016-08-26:
@geky The issue is still valid. The problem does not arise if you are using Nanostack binaries with public release. However, if you are using sources file name conflict will arise in Windows OS. Because Windows is case insensitive. Javier explained it very well above.

sg-, 2016-08-26:
This shouldn't be the case. mbed.h (rtos.h included by mbed.h) is the entry point and should use path referenced includes. If this is not the case, please open an issue for that. Its not clear here what you're saying the problem is.

hasnainvirk, 2016-08-26:
@kjbracey-arm Can you please explain here what we saw while compiling with Nanostack sources ?
I tried to explain inthis comment https://github.com/ARMmbed/mbed-os/issues/2411#issuecomment-242642831 what I understood. If you have a better explanantion, please provide.

sg-, 2016-08-26:
@javier-moreno-tridonic-com it sounds like the main function is missing. The original issue is confirmed to be resolved. If there is a different issue, please open it with steps to reproduce.

closed 2016-08-26 by sg-