ARMmbed/mbed-os #2412: does not recompile when command line changed.

LiyouZhou, 2016-08-10:
I added `-o debug-info` to `mbed compile` It did not recompile things.

0xc0170, 2016-08-12:
cc @theotherjimmy @screamerbg @bridadan

bridadan, 2016-08-12:
I'm pretty sure even for systems like `make` you need to run a "clean" build when switching between debug and release builds. The only way to know if you'd need to do a clean build is store what the previous `-o` arguments were and clean if they are different. Do we really want to do that?

LiyouZhou, 2016-08-12:
It is not a bug but a feature. `make` is pretty bare bone, we should aim to provide at least some features that make doesn't. It's not difficult to store a hash of the command line. This is similar to detecting if the source file have changed, or if the config file have changed. I argue the user could expect the tool to do the right thing and it would be nice if the tool actually did. I agree it might not be a `must`, but a `nice to have`.

bridadan, 2016-08-12:
@LiyouZhou Thanks clarifying, you're right it would be awesome if the tools took care of this.

@0xc0170 any chance we can update the label?

theotherjimmy, 2017-05-16:
see https://github.com/ARMmbed/mbed-os/pull/4331

PR #4331 (2017-05-26): Have objects depend on their respective compiler invocations
# Description

Each of C, C++, ASM and linker now depend on their command line invocations.
For example, adding `-DFoo` to the mbed CLI invocation (`mbed compile -DFoo`)
would force a complete recompile. Further, switching build profiles will force
all C and C++ to recompile. The assembler flags are the same ¯\\\_(ツ)\_/¯.

Resolves #2412 

# TODO

- [x] manual verification that modifying each part of a build profile causes
  the appropriate rebuild
- [x] /morph test
closed 2017-05-26 by sg-