ARMmbed/mbed-os #2420: Maximum number of open sockets

tommikas, 2016-08-11:
It seems that currently the maximum total number of simultaneously available sockets is 4, in any combination of TCPServer, TCPSocket and UDPSocket instances. UDPSocket is further limited to a maximum of 2 (4 UDP sockets total, but one seems to be reserved for DNS and one for DHCP, leaving 2 for the application).

Are these limits documented anywhere? Are they configurable?

infinnovation, 2016-08-11:
I think this is the relevant definition in features/net/FEATURE_IPV4/lwip-interface/lwip/include/lwip/opt.h

```C
/**
 * MEMP_NUM_NETCONN: the number of struct netconns.
 * (only needed if you use the sequential API, like api_lib.c)
 */
#ifndef MEMP_NUM_NETCONN
#define MEMP_NUM_NETCONN                4
#endif
```

0xc0170, 2016-08-12:
cc @geky

geky, 2016-08-12:
Looking at the referenced issue, you can only make 4 sequentially? That sounds like a memory-leak that should be looked into, or is it in a different issue?

@infinnovation is right, the `MEMP_NUM_NETCONN` can be overridden for additional sockets (at a memory cost). However, _reserving_ a socket for DNS/DHCP sounds wrong, that may be a loss from the above memory-leak.

tommikas, 2016-08-15:
>Looking at the referenced issue, you can only make 4 sequentially? That sounds like a memory-leak that should be looked into, or is it in a different issue?

I don't think it's a memory leak, since I can free instances and recreate them in any order as many times as I wish. I.e. it's not a limitation on how many I can make, only on how many can exist simultaneously.

>@infinnovation is right, the MEMP_NUM_NETCONN can be overridden for additional sockets (at a memory cost). 

Alright. This might be worth documenting somewhere. It takes quite a bit of digging to find as it is.

>However, reserving a socket for DNS/DHCP sounds wrong, that may be a loss from the above memory-leak.

To be precise, during EthernetInterface::connect() [udp_new()](https://github.com/ARMmbed/mbed-os/blob/master/features/net/FEATURE_IPV4/lwip-interface/lwip/core/lwip_udp.c#L930) gets called twice, once from [dhcp_start()](https://github.com/ARMmbed/mbed-os/blob/master/features/net/FEATURE_IPV4/lwip-interface/lwip/core/lwip_dhcp.c#L674) and once from [dns_init()](https://github.com/ARMmbed/mbed-os/blob/master/features/net/FEATURE_IPV4/lwip-interface/lwip/core/lwip_dns.c#L245). Thus 2 of the available 4 udp PCBs are reserved.

[udp_remove()](https://github.com/ARMmbed/mbed-os/blob/master/features/net/FEATURE_IPV4/lwip-interface/lwip/core/lwip_udp.c#L899) doesn't get called at all, so I guess that could be a memory leak? Although I could see DNS needing a guaranteed available socket...

Calling EthernetInterface::disconnect() releases the PCB reserved by dhcp. Since it's currently possible to create sockets on a disconnect interface, that makes it possible to open a 3rd UDPSocket instance, (which then of course makes a new EthernetInterface::connect() fail because it can't get a new PCB).

(As a side note, I wonder what happens/should happen to existing sockets when the network interface is disconnected... )

tommikas, 2016-08-23:
http://jira.arm.com/browse/IOTSFW-2861

sg-, 2016-08-25:
This needs to be defined as the minimum guaranteed number of sockets per interface and documented.

geky, 2016-11-02:
Does https://github.com/ARMmbed/mbed-os/pull/3078 resolve this?

tommikas, 2016-11-03:
Mostly this was just about these restrictions being documented somewhere, so yes, this helps. It's still not very visible, but at least it's something.

closed 2016-11-03 by tommikas