ARMmbed/mbed-os #2421: Socket::open() behaviour with no network connectivity

tommikas, 2016-08-11:
Calling Socket::open() before EthernetInterface::connect() results in `sys_arch_protect error`.
As a NetworkInterface instance is required by sockets ([as documented](https://docs.mbed.com/docs/mbed-os-api-reference/en/5.1/APIs/communication/network_sockets/#the-networkinterface-classes)), I wouldn't necessarily expect it to work (although possibly not with that specific error).

But, as long as EthernetInterface::connect() has been called once, even if DHCP fails or connect() is immediately followed by EthernetInterface::disconnect(), I can then open sockets as if I had working network connectivity.

Is this the intended behaviour in each situation?

0xc0170, 2016-08-12:
cc @geky

geky, 2016-08-12:
Thanks for reporting this. You are right, this error should be reported more consistently (perhaps with `NSAPI_ERROR_PARAMETER`). Opening sockets on a disconnect interface should fail as well.

tommikas, 2016-08-15:
Sounds good to me.

tommikas, 2016-08-23:
http://jira.arm.com/browse/IOTSFW-2862

geky, 2016-08-26:
This should be fixed on https://github.com/ARMmbed/mbed-os/pull/2561, let me know if there's stlil and issue

sg-, 2016-09-19:
@tommikas Is this resolved?

tommikas, 2016-09-20:
Yep. Just been waiting for the changes to land on master. Closing now.

closed 2016-09-20 by tommikas