ARMmbed/mbed-os #2424: Rebinding TCPServer socket more than once results in an infinite loop

tommikas, 2016-08-11:
1. create and open a TCPServer socket
2. bind the socket to port X
3. bind the same socket again to port Y
4. bind any TCPServer type socket to port Z

Step 4 always results in an infinite loop in [lwip_tcp.c : lines 425-426](https://github.com/ARMmbed/mbed-os/blob/master/features/net/FEATURE_IPV4/lwip-interface/lwip/core/lwip_tcp.c#L425).

0xc0170, 2016-08-12:
cc @geky

tommikas, 2016-08-23:
http://jira.arm.com/browse/IOTSFW-2863

sg-, 2016-08-25:
Please file these issues with a test case or example program or code snippet to reproduce.

geky, 2016-08-29:
Looking into this more, it appears lwip does not handling rebinding sockets.

In BSD sockets, bind is similarly not allowed more than once:
http://man7.org/linux/man-pages/man2/bind.2.html

Fortunately, the state of the socket can be determined from the netconn struct, so changing the behaviour to return `NSAPI_ERROR_PARAMETER` is preferential to a non-halt state.

geky, 2016-08-29:
Added a check per the above behaviour here: https://github.com/ARMmbed/mbed-os/pull/2580

Let us know if there is still an issue.

tommikas, 2016-08-30:
>changing the behaviour to return NSAPI_ERROR_PARAMETER is preferential to a non-halt state

Yep, that sounds good. I'll try it out.

tommikas, 2016-09-14:
This has been fixed. Closing.

closed 2016-09-14 by tommikas