ARMmbed/mbed-os #2435: Socket::set_timeout doesn't work properly for the server side TCPClient instance

tommikas, 2016-08-12:
1. create, open, bind and set to listen a new TCPServer instance
2. accept a new connection on a new TCPSocket instance
3. set the new TCPSocket to non-blocking mode
4. call TCPSocket::recv

TCPSocket::recv blocks indefinitely if no data is pending or received.

geky, 2016-08-22:
Hi @tommikas, thanks for reporting this issue. It looks like this has been fixed by the unrelated pr from @c1728p9: https://github.com/ARMmbed/mbed-os/pull/2502

Let us know if the problem still persists : )

tommikas, 2016-08-23:
http://jira.arm.com/browse/IOTSFW-2864

sg-, 2016-08-25:
Please file these issues with a test case or example program or code snippet to reproduce.

Can you confirm if #2502 resolved this issue?

tommikas, 2016-09-01:
#2502 fixed this. Closing.

closed 2016-09-01 by tommikas