ARMmbed/mbed-os #2438: Socket:: bind() behaviour is inconsistent

tommikas, 2016-08-12:
1. A UDPSocket type socket can bind to the same port repeatedly.
2. Two UDPSocket type sockets cannot bind to the same port at the same time.

3. A TCPServer type socket cannot bind to the same port repeatedly.
4. Two TCPServer type sockets cannot bind to the same port at the same time.

5. A TCPServer type socket and a UDPSocket type socket can bind to the same port at the same time. After that each socket behaves as described previously.

Mostly the last one seems like it could potentially cause issues.

infinnovation, 2016-08-12:
> A TCPServer type socket and a UDPSocket type socket can bind to the same port

This is expected behaviour.  TCP port numbers and UDP port numbers are in different number-spaces; they are different protocols.

geky, 2016-08-12:
> 2. Two UDPSocket type sockets cannot bind to the same port at the same time.
> 4. Two TCPServer type sockets cannot bind to the same port at the same time.

This sounds expected, otherwise one socket would get packets intended for the other. This could corrupt transactions from an application that assumes it owns the data stream.

> 5. A TCPServer type socket and a UDPSocket type socket can bind to the same port at the same time. After that each socket behaves as described previously.

This is expected as described by @infinnovation.

> 2. Two UDPSocket type sockets cannot bind to the same port at the same time. 
> 4. Two TCPServer type sockets cannot bind to the same port at the same time.

I'm currently looking into this, I suspect lwip has a sort of `REUSEADDR` setting it is enforcing for the same reasons as preventing two sockets to bind to the same port. However, it is odd that this is only affecting the TCP socket.

geky, 2016-08-13:
> 2. Two UDPSocket type sockets cannot bind to the same port at the same time.
> 4. Two TCPServer type sockets cannot bind to the same port at the same time.

Looking into it more, this is a part of a similar mechanism that prevents sockets from binding to the same port as a part of protecting a user from himself. The fact that UDP bind succeeded is curious, but I suspect just an idiosyncrasy of lwip's implementation.

Once #2440 is merged, you can use NSAPI_REUSEADDR to allow socket servers to reuse the same address without waiting.

tommikas, 2016-08-15:
>This is expected behaviour. TCP port numbers and UDP port numbers are in different number-spaces; they are different protocols.

Ah of course. That's no problem then. Thanks.

>Looking into it more, this is a part of a similar mechanism that prevents sockets from binding to the same port as a part of protecting a user from himself. The fact that UDP bind succeeded is curious, but I suspect just an idiosyncrasy of lwip's implementation.

Fair enough.

>Once #2440 is merged, you can use NSAPI_REUSEADDR to allow socket servers to reuse the same address without waiting.

Cool. I wondered if this might be connected to #2408, but seemed different enough to create its own issue.

closed 2016-08-23 by tommikas