ARMmbed/mbed-os #2439: calling eth.connect() on the Ethernet interface causes an NSAPI_ERROR_DHCP_FAILURE to be returned

LiyouZhou, 2016-08-12:
mbed-os 5.1.0

Expected behaviour `return success if already connected`

geky, 2016-08-12:
Thanks for reporting this issue.

@sg-, @c1728p9, what do you think, should calling connect on an interface multiple times be defined? Alternatively it could be modified to return NSAPI_ERROR_PARAMETER to make it clear the problem is with how the function is called.

sg-, 2016-08-12:
@geky how would you propose someone determine if connected to a network (not to the internet)? There doesn't seem to be much precedent here.

geky, 2016-08-12:
Currently a user can check the return value of connect to know if a connection was succesful. A user can also check `get_ip_address` to tell if DHCP succeeded, but this will only work until we have static IP addresses.

If we want to store the state of the network connectivity, we could add an `is_connected` function. I believe the idea came up during prototyping the network-socket API, although I'm not sure why it was removed.

sg-, 2016-08-12:
I thought about that but what would ```is_connected check```? Totally opposed to state variables for many reasons but in this case given that if the network rejected or kicked your node off it would return a false state.

geky, 2016-08-13:
> Totally opposed to state variables for many reasons

Same. I'm worried it may give the user a false sense of expectations since most ports probably can't detect if the network disconnects. I checked and lwip doesn't seem to have a system for reporting this sort of state.

At the same time, if multiple user's code is calling `connect`, who decides what the correct configuration is? The `connect` call with no arguments wouldn't work for wifi interfaces.

I'm of the opinion it should be up to the user to store the state if they need it, and for user libraries they should expect any provided network interfaces to have been brought up. If errors are appropriately propagates from socket calls, we can insure the errors reflect that the network was unavailable. I'm curious to see other thoughts though.

sg-, 2016-08-13:
> I'm of the opinion it should be up to the user to store the state if they need it, and for user libraries they should expect any provided network interfaces to have been brought up. If errors are [...]

Agree - @LiyouZhou can you share how or what you'd expect the network to be checking for an existing or established connection?

LiyouZhou, 2016-08-17:
>for user libraries they should expect any provided network interfaces to have been brought up.

This is what I care about. It would be nice if I didn't have to depend on the user and only connect the network when it is strictly necessary. There is also a documentation overhead where I have to remind users to connect the network before spinning up the library. There is also overhead on the user to figure out which web interface is used in the library and the correct API to call to spin it up. It is fine if you decided to go the other way. But please make this point clear in documentation somewhere for library developers and application developers alike.

Now back to the original point: What should you return when connect is called the second time?  Your current answer is "behavior undefined". I argue that many people will try to do this, regardless of weather they should or not.  You should define a behavior in these cases and document this.

I'm not very familiar with the mechanism of ethernet and maybe `NSAPI_ERROR_DHCP_FAILURE` is the right thing to return. But I did not expect and error because it was not how the interface worked before. I think what I want is "define a behavior and commit to it".

geky, 2016-08-26:
#2561 changes the behaviour to return `NSAPI_ERROR_PARAMETER` to indicate that connecting an already connected network interface is invalid.

Allowing the network to be reconnected multiple times is not a robust solution to the library writer's problem, since it doesn't consider other types of network interfaces, such as WiFi for cellular, where connecting for the user removes control of the interface (such as ssid/carrier) from the user.

The best behaviour a library can do when a network is disconnected is to convey that information to the user in the form of an error. If, for instance, a user passes in a null EthernetInferface, the library will be simply unable to bring up the network.

sg-, 2016-09-19:
@LiyouZhou Is this resolved?

LiyouZhou, 2016-09-19:
From @geky 's previous comment, it looks like it is.

closed 2016-09-19 by sg-