ARMmbed/mbed-os #2444: Some boards have inverted built-in LED state, this should be solved by HAL

janjongboom, 2016-08-15:
Every example that we write that targets multiple microcontrollers needs to deal with the fact that the wiring of built-in LEDs differs between micros. Sometimes you need to turn the GPIO port off to turn the LED on, other times you need to turn the port on. This makes writing portable code a pain in the ass, as every first instruction needs to be: 'check how the built-in LEDs are wired, and then declare some macro'. That is not good, but should be part of the HAL.

Suggestion: Add two new defines, `BUILT_IN_LED_ON` and `BUILT_IN_LED_OFF`. Default them to 1/0, and allow overriding for some targets (like K64F) to 0/1.

cc @niklas-arm @theotherjimmy @matthewelse

bridadan, 2016-08-15:
I thought we had this at one point? Something like `HIGH` or `LED_ON`?

janjongboom, 2016-08-16:
I see that some targets have this yeah, f.e. Maxim targets have an enum defined with LED_ON / LED_OFF. Unfortunately that's not portable.

bridadan, 2016-08-16:
@janjongboom I agree. Perhaps we could prefix this with the API we're trying to target? Ex. `DIGITAL_ON`/`DIGITAL_OFF` or `DIGITAL_HIGH`/`DIGITAL_LOW` ?

jacobrosenthal, 2016-08-16:
Something I didnt originally think about with inverted leds, PWM is kind of
a nightmware as well, as 100% on isnt really a good way of turning things
off, and there are multiple tickets about how it is currently not possible
to disable pwm from mbed.

janjongboom, 2016-08-16:
@jacobrosenthal Yeah, I agree, although I guess it's a different issue than this. However, it is a problem as currently I have to write code like https://github.com/ARMmbed/connected-lights/blob/master/lighting-system-firmware/source/config.h#L18 to make stuff portable.

However, that would involve more work... Maybe move that to a separate issue.

@bridadan I think we should have BUILT_IN (or BUILTIN) as part of the name, as it'll only apply to those LEDs. Everything else will be wired by developer itself.

jacobrosenthal, 2016-08-16:
Yeah NP, not trying to hijack. But +1 to simplifying my handling of
inverting digital outputs.

On Tue, Aug 16, 2016 at 12:47 PM, Jan Jongboom <notifications@github.com>
wrote:

> olve more work... Maybe move that to a separate issue.
>
> @bridadan <https://github.com/bridadan> I think we should have BUILT_IN
> (or BUILTIN) as part of the name, as it'll only apply to those LEDs.
> Everything else will be wired b
>

rimomaguiar, 2017-06-11:
Use the chip 74LS04 hexa-inverter to invert the LED status, pin 7 is negative, pin 14 is positive, pin 1 = IN, pin 2 = OUT INVERTED, pin 3 = IN, pin 4 = OUT INVERTED... it cost is about 60 cent per chip.

janjongboom, 2018-12-19:
Why is this closed.

closed 2019-10-31 by ciarmcom