ARMmbed/mbed-os #2446: nRF51822+S130+HRSensor= disconnection

gaucho1978, 2016-08-15:
I'm using nRF51822 (RedBear Nano board) with MBED library that uses S130 Nordik library.

There is an implementation that let's me use mbed library over arduino:

I use the nRF51822  simoultaeously as master role (in order to connect a HR sensor, a Polar H7 HRS) and as peripherial role (to let tablets to connect to my custom device).

This is what I do:

1) I perform a BT scan and I start a Connection from a tablet and then I successfully start to receive notifications from nRF51822 viewing them on the tablet log.

2) I Perform a BT scan from nRF51822 and it finds the HR device (at this point everything continues to work correctly)

3) I perform a BT connection from nRF51822 to the HR sensor and i register for notifications  and i start to receive Heart Rate Data from the HRS, viewing them on a serial print of the RedBearNano. At the same moment the notifications stops to be received on the tablet. 

I can't understand why this happens. In order to start to receive again the notifications on the tablet, i must request, from the tablet, a disconnection of the peripherial (the nRF51822) and then request again a Connection and register again for notifications.

Looking on the sniffed packets, it seems there is nothing strange (because I also own a bluefruit sniffer).

I add 2 important links to my previous message: 

1) the MBED + Arduino library that I am using (same library as here): https://github.com/RedBearLab/nRF51822-Arduino

2) the code that I am running for this DEBUG (here annexed)
[treadmill_nano.zip](https://github.com/ARMmbed/mbed-os/files/417857/treadmill_nano.zip)

0xc0170, 2016-08-15:
cc @pan-

gaucho1978, 2016-08-19:
Can you guide me to check the variables responsible for the problem?

the routine that sends the notifications to the tablet is the updateCharacteristicValue that you can find in my code. that routine corresponds to the write() routine in https://github.com/ARMmbed/mbed-os/blob/db60d6a7f72613a53df4a19b3d73849819013680/features/FEATURE_BLE/targets/TARGET_NORDIC/TARGET_MCU_NRF51822/source/nRF5xGattServer.cpp (see from line 212).

this write function calls the NORDIC sd_ble_gatts_hvx function .

when the call goes fine it returns 0. When the problem occours it returns ERROR_BLEGATTS_SYS_ATTR_MISSING.

How could I check variables to understand what is going wrong?

pan-, 2016-08-19:
In your code you don't use the connection handle to update the characteristic value, if you use the handle of the connection between the tablet and the NRF51822, this might help.

```c++
void publishOverSerialBT() {
  // some code ..
  uint16_t updateCharValResult= ble.updateCharacteristicValue(
    connectionRunsocialHandle,  characteristic2AttributeHandle, rx_buf, rx_buf_num
  );
  // rest of the code
}
```

gaucho1978, 2016-08-19:
WOW. IT WORKS!

gaucho1978, 2016-08-19:
you saved my life. thank you very much! I love you.

closed 2016-08-19 by gaucho1978