ARMmbed/mbed-os #2448: [exporter] LPCXpresso exporter broken

toyowata, 2016-08-15:
Exported project for LPCXpresso broken from mbed library Rev.122 and onward.

## Step to reproduce:

1. Create blinky project for LPC1768 (or other LPC) target
2. Set mbed library Rev. 122 or 123
3. Export to LPCXpresso
4. Open the project by LPCXpresso IDE (I used LPCXpresso v8.1.4)
5. Build the project file

```
Building target: mbed_blinky.axf
Invoking: MCU C++ Linker
arm-none-eabi-c++ -nostdlib -L"/Users/toyowata/Documents/LPCXpresso_8.1.4/workspace/mbed_blinky" -L"/Users/toyowata/Documents/LPCXpresso_8.1.4/workspace/mbed_blinky/mbed" -L"/Users/toyowata/Documents/LPCXpresso_8.1.4/workspace/mbed_blinky/mbed/TARGET_LPC1768" -L"/Users/toyowata/Documents/LPCXpresso_8.1.4/workspace/mbed_blinky/mbed/TARGET_LPC1768/TARGET_NXP" -L"/Users/toyowata/Documents/LPCXpresso_8.1.4/workspace/mbed_blinky/mbed/TARGET_LPC1768/TARGET_NXP/TARGET_LPC176X" -L"/Users/toyowata/Documents/LPCXpresso_8.1.4/workspace/mbed_blinky/mbed/TARGET_LPC1768/TARGET_NXP/TARGET_LPC176X/TARGET_MBED_LPC1768" -Xlinker -Map="mbed_blinky.map" -Xlinker --gc-sections -mcpu=cortex-m3 -mthumb -T "" -o "mbed_blinky.axf"  ./main.o   
/Applications/lpcxpresso_8.1.4_606/lpcxpresso/tools/bin/../lib/gcc/arm-none-eabi/5.2.1/../../../../arm-none-eabi/bin/ld: read in flex scanner failed
collect2: error: ld returned 1 exit status
make: *** [mbed_blinky.axf] Error 1
```

When I use Rev. 121 mbed library, it worked fine (I could build a project by LPCXpresso).  The exporter doesn't seem to generate pre-build object files for target.  See below:

## mbed library Rev.122 and onward
[image]

## mbed library Rev.121 and upward
[image]

I also tried to export other target such as LPC1114FN28 anr LPCXpresso11U68, they are same result.

0xc0170, 2016-08-15:
Can you test the latest master here with ``project.py`` to find out if its a problem of scripts here in this code base or in the online IDE?

toyowata, 2016-08-19:
@0xc0170 

```tools\project.py -m LPC1768 -i lpcxpresso -p 0``` generates correct project file for the LPCXpresso IDE.  I got link errors, and fixed by other PR (see later - Other issues).

```tools\project.py -m LPC1768 -i lpcxpresso -p 0 -b``` it seems generate incorrect files in the project zip.  I got compile errors below:

```
make -r all 
Building file: ../zip/LPC1768/temp/rtos/rtx/TARGET_CORTEX_M/TARGET_M3/TOOLCHAIN_ARM/HAL_CM3.c
Invoking: MCU C Compiler
arm-none-eabi-gcc -D__NEWLIB__ -D__CODE_RED -DCPP_USE_HEAP -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\emblocks" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\emblocks\LPC1768" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\emblocks\LPC1768\temp" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\emblocks\LPC1768\temp\env" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\emblocks\LPC1768\temp\rtos" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\emblocks\LPC1768\temp\rtos\rtos" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\emblocks\LPC1768\temp\rtos\rtx" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\emblocks\LPC1768\temp\rtos\rtx\TARGET_CORTEX_M" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\emblocks\LPC1768\temp\rtos\rtx\TARGET_CORTEX_M\TARGET_M3" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\emblocks\LPC1768\temp\rtos\rtx\TARGET_CORTEX_M\TARGET_M3\TOOLCHAIN_GCC" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\env" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\hal" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\hal\api" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\hal\common" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\hal\hal" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\hal\hal\storage_abstraction" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\hal\targets" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\hal\targets\cmsis" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\hal\targets\cmsis\TARGET_NXP" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\hal\targets\cmsis\TARGET_NXP\TARGET_LPC176X" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\hal\targets\cmsis\TARGET_NXP\TARGET_LPC176X\TOOLCHAIN_GCC_CR" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\hal\targets\hal" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\hal\targets\hal\TARGET_NXP" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\hal\targets\hal\TARGET_NXP\TARGET_LPC176X" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\hal\targets\hal\TARGET_NXP\TARGET_LPC176X\TARGET_MBED_LPC1768" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\mbed" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\mbed\TARGET_LPC1768" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\mbed\TARGET_LPC1768\TARGET_NXP" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\mbed\TARGET_LPC1768\TARGET_NXP\TARGET_LPC176X" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\mbed\TARGET_LPC1768\TARGET_NXP\TARGET_LPC176X\TARGET_MBED_LPC1768" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\mbed\TARGET_LPC1768\TOOLCHAIN_GCC_CR" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\zip" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\zip\LPC1768" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\zip\LPC1768\temp" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\zip\LPC1768\temp\env" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\zip\LPC1768\temp\rtos" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\zip\LPC1768\temp\rtos\rtos" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\zip\LPC1768\temp\rtos\rtx" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\zip\LPC1768\temp\rtos\rtx\TARGET_CORTEX_M" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1\zip\LPC1768\temp\rtos\rtx\TARGET_CORTEX_M\TARGET_M3" -I"C:\Users\toywat01\Documents\LPCXpresso_8.0.0_526\workspace\MBED_A1" -O0 -g3 -Wall -c -std=gnu99 -DTARGET_MBED_LPC1768 -DDEVICE_ERROR_PATTERN=1 -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DTARGET_NXP -DTARGET_LPC176X -DDEVICE_PORTIN=1 -DDEVICE_RTC=1 -DTOOLCHAIN_object -D__CMSIS_RTOS -DMBED_BUILD_TIMESTAMP=1471585027.01 -DTOOLCHAIN_GCC -DDEVICE_CAN=1 -DTARGET_LIKE_CORTEX_M3 -DTARGET_CORTEX_M -DARM_MATH_CM3 -DDEVICE_ANALOGOUT=1 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_M3 -DDEVICE_PWMOUT=1 -DDEVICE_INTERRUPTIN=1 -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -D__CORTEX_M3 -DDEVICE_STDIO_MESSAGES=1 -DTARGET_LPC1768 -DTOOLCHAIN_GCC_CR -DDEVICE_PORTINOUT=1 -DDEVICE_SERIAL_FC=1 -DTARGET_RELEASE -D__MBED_CMSIS_RTOS_CM -DDEVICE_SLEEP=1 -DDEVICE_SPI=1 -DDEVICE_ETHERNET=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_SERIAL=1 -DDEVICE_SEMIHOST=1 -DDEVICE_DEBUG_AWARENESS=1 -DDEVICE_LOCALFILESYSTEM=1 -include mbed_config.h -c [... cut ...]
| OK      | LPC1768 | GCC_CR    | RTOS_8      | ISR (Queue)                           |        7.81        |       20      |  1/1  |
+---------+---------+-----------+-------------+---------------------------------------+--------------------+---------------+-------+
Result: 8 OK / 21 TIMEOUT

Completed in 732.83 sec

```

It failed all non-RTOS test cases.  This is because the software_init_hook is always enabled even if non-RTOS build here.
https://github.com/ARMmbed/mbed-os/blob/master/hal/targets/cmsis/TARGET_NXP/TARGET_LPC176X/TOOLCHAIN_GCC_CR/startup_LPC17xx.cpp#L154
I will find out this issue.

toyowata, 2016-08-30:
I have created other ticket for automated test fail issue here : https://github.com/ARMmbed/mbed-os/issues/2581

sg-, 2016-09-19:
@toyowata Is this resolved with the latest exporter updates?

toyowata, 2016-09-20:
@sg- Actually, it is getting worse.  I got an error dialog below when I exported to the LPCXpresso.

[image]

### Step to reproduce

1. Create new blinky program for NXP LPC target such as LPC1768
2. Export the project to LPCXpresso

theotherjimmy, 2016-09-20:
reproduced. I have the error message at hand. Could you try exporting from the command line tools?

that is:
```
mbed import mbed-os-blinky
mbed export -i lpcxpresso -m lpc1768
```
then import the project into lpcxpresso

This would help, as the error you are seeing is only related to how the online export system works. If that's the only issue, then exporting from the command line should work.

toyowata, 2016-09-21:
@theotherjimmy 

The mbed export command generates LPCXpresso project without error.  However, I got following error when I build the project by LPCXpresso IDE.

```
**** Build of configuration Debug for project Unnamed_Project ****

Nothing to build for project Unnamed_Project
```

The project windows doesn't seem to have source code and libraries.

[image]

JojoS62, 2016-09-21:
I've tried also the mbed-cli with the blinky example. The export stops after scanning FEATURE_STORAGE and the exported project file is not generated.

```
D:\Projects\Sn\ARM\LPCXpresso1769\mbed-os-example-blinky>pip install mbed-cli --upgrade
Requirement already up-to-date: mbed-cli in c:\python27\lib\site-packages\mbed_cli-0.9.5-py2.7.egg

D:\Projects\Sn\ARM\LPCXpresso1769\mbed-os-example-blinky>mbed export -i lpcxpresso -m lpc1768 -vv
[mbed] Working path "D:\Projects\Sn\ARM\LPCXpresso1769\mbed-os-example-blinky" (program)
[mbed] Exec "python -u D:\Projects\Sn\ARM\LPCXpresso1769\mbed-os-example-blinky\mbed-os\tools\project.py -i lpcxpresso -
m lpc1768 --source ." in D:\Projects\Sn\ARM\LPCXpresso1769\mbed-os-example-blinky
Build Options: debug-info
Scan: .
Scan: FEATURE_BLE
Scan: FEATURE_CLIENT
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_UVISOR
Scan: FEATURE_IPV4
Scan: FEATURE_IPV6
Scan: FEATURE_STORAGE

D:\Projects\Sn\ARM\LPCXpresso1769\mbed-os-example-blinky>
```

I can compile the the project with GCC_ARM. With GCC_CR, the .bin is created also but there is an error when the size should be prompted:
`
Link: mbed-os-example-blinky
Elf2Bin: mbed-os-example-blinky
Unknown toolchain for memory statistics GCC_CR
Image: .\.build\lpc1768\gcc_cr\mbed-os-example-blinky.bin
`

JojoS62, 2016-09-21:
Update: I was looking for an archive file in .build, but a .project and .cproject for the eclipse ide is built in the root of my blinky-example project. These files look ok, but when I import from the root of my project, the import pulls in all the unwanted stuff like the complete mbed-os dir with all tools etc.
[Uploading mbed-os-example-blinky.zip…]()

theotherjimmy, 2016-09-21:
@toyowata what version of mbed-os are you using?

@JojoS62 Comment 1: That error is likely related to memap.py, could you open a new issue for that link error so that we can track it separately?

@JojoS62 Comment 2: The .project and .cproject file in the root is expected; zip files are only created for mbed 2 tests. The source of mbed-os (*.c, *.cpp) should be pulled in to the project, but the tools should not be in the project workspace. Further, I can't seem to download the zip file you posted.

toyowata, 2016-09-23:
@theotherjimmy I used mbed-os-5.1.4 for [this](https://github.com/ARMmbed/mbed-os/issues/2448#issuecomment-248511559) test and rev.126 for [this](https://github.com/ARMmbed/mbed-os/issues/2448#issuecomment-248167360).

I also found some reports from community questions below:
https://developer.mbed.org/questions/74489/Exprt-error-LPCxpresso/
https://developer.mbed.org/questions/74741/Exporting-EA-LPC11U35-Quickstart-board-t/

Carl-M, 2016-10-03:
I have this issue too, started new mbed project, choose blinky and export from online compiler. Import into lpcxpresso v8.2.2 [Build 650]  and get the same -t error.
If I switch to revision 121 and repeat it builds fine in LPCxpresso

theotherjimmy, 2016-10-03:
@Carl-M I'm sorry, but I don't know what you are talking about when you refer to the "-t error"

Carl-M, 2016-10-03:
Sorry this is the error 
`arm-none-eabi-c++ -nostdlib -L"\LPCXpresso_8.2.2_650\workspace\mbed_blinky" -L"\LPCXpresso_8.2.2_650\workspace\mbed_blinky\mbed" -L"\LPCXpresso_8.2.2_650\workspace\mbed_blinky\mbed\TARGET_LPC1768" -L"\LPCXpresso_8.2.2_650\workspace\mbed_blinky\mbed\TARGET_LPC1768\TARGET_NXP" -L"\LPCXpresso_8.2.2_650\workspace\mbed_blinky\mbed\TARGET_LPC1768\TARGET_NXP\TARGET_LPC176X" -L"\LPCXpresso_8.2.2_650\workspace\mbed_blinky\mbed\TARGET_LPC1768\TARGET_NXP\TARGET_LPC176X\TARGET_MBED_LPC1768" -Xlinker -Map="mbed_blinky.map" -Xlinker --gc-sections -mcpu=cortex-m3 -mthumb -T "" -o "mbed_blinky.axf"  ./main.o   
e:/dev/nxp/lpcxpresso_8.2.2_650/lpcxpresso/tools/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld.exe: unrecognized option '-T'
e:/dev/nxp/lpcxpresso_8.2.2_650/lpcxpresso/tools/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld.exe: use the --help option for usage information`

theotherjimmy, 2016-10-03:
You don't have a linker script. It should be between the `"`s after the `-T`

theotherjimmy, 2016-10-03:
What target did you try?

Carl-M, 2016-10-03:
Mbed LPC1769

theotherjimmy, 2016-10-03:
Do you mean 1768?

JojoS62, 2016-10-03:
The -t is a linker option that is wrong. I experienced also that options are defined double, once in project settings and also in additional misc settings.

theotherjimmy, 2016-10-03:
@JojoS62 @Carl-M I appreciate you enthusiasm for this exporter.

 The exporter in question is called codered in the tools (file [here](https://github.com/ARMmbed/mbed-os/blob/master/tools/export/codered.py)) In the current master, once issue #2892 Is fixed, you may use the eclipse_gcc_arm or similar exporter to have an equivalent export. That is the way we have planned forward. 

OTOH, if you would like to dive into the tools code yourself, I can help you with that.

JojoS62, 2016-10-03:
Thanks, eclipse exporter is what I appreciate also. Lpcxpresso uses the GCC toolchain, do no need for a different handling. I see only differences in the linker script that is generated by lpcxpresso. Using the gcc_arm scripts work, but switching to lpcxpresso managed scripts makes trouble. 
I'm wiling to investigate this further, a compatible solution would be great.

theotherjimmy, 2016-10-03:
>Using the gcc_arm scripts work, but switching to lpcxpresso managed scripts makes trouble.

Perfect! Is there any special handling that LPCXpresso needs? (Special flashing tools, Code analysis tools, etc?)

fjvva, 2016-10-05:
In order to allow code to compile, a quick fix is ->Project->Properties->Settings->Tool Settings->MCU C++ linker->Managed Linker Script
And then just check the box "Manage Linker Script"

This worked for me

theotherjimmy, 2016-10-20:
@JojoS62 It's been a few weeks! If you could get back to me on this:

>Is there any special handling that LPCXpresso needs? (Special flashing tools, Code analysis tools, etc?)

Then I can get on making the LPCXpresso exporter more robust.

JojoS62, 2016-10-21:
thanks for asking. I'm still struggling with the exporters in general. I use the mbed-os master branch and I execute the following:
```
D:\Projects\Sn\LocalGit\mbed-os\tools>python project.py -n MBED_BLINKY -i lpcxpresso
Scan: blinky
Scan: env
Scan: hal

D:\Projects\Sn\LocalGit\mbed-os\tools>
```
That gives me an output in mbed-os\export where still a lot is missing and the dir structure is messed up, even a mbed.h. I have attached the complete folder 'MBED_BLINKY_lpcxpresso_LPC1768 '
[MBED_BLINKY_lpcxpresso_LPC1768.zip](https://github.com/ARMmbed/mbed-os/files/543735/MBED_BLINKY_lpcxpresso_LPC1768.zip)
So, am I doing something wrong here or does the export work now only with mbed-cli?

For the lpcxpresso most things worked before the directory structure change. The other issue was double output of flags and missing/wrong settings in the .cproject file. I had opened an issue for that but accidently closed it. I can reopen it and I will post my fix for the exporter there.

theotherjimmy, 2017-04-25:
@toyowata This exporter has not received the love that it should have.  I don't have time right now to fix the issues with the LPCXpresso exporter, but if you do I have some guidance.  I would start by trying the gnuarmeclipse exporter in place of lpcxpresso.  If that is satisfactory, then you can just change the `lpcxpresso` key in `tools/export/__init__.py` to point to the same thing as the `gnuarmeclipse` in the `EXPORTERS` dict. If the replacement does not look satisfactory, then we can discuss extending the class.

JojoS62, 2018-02-16:
I suggest to close this issue, LPCXpresso has been superseeded by MCUXpresso and projects can be exported to MCUX.

closed 2018-02-16 by 0xc0170