ARMmbed/mbed-os #2449: icmp/raw socket support

tommikas, 2016-08-15:
The network-socket API doesn't seem to support either at the moment.
Would it be possible to add it, so for example ping could be implemented on top of the network-socket API?

0xc0170, 2016-08-15:
cc @geky

geky, 2016-08-15:
We accept patches ; )

This would be a good feature to have, but at the moment we are focusing on cementing the current network-socket API with minimal modifications.

tommikas, 2016-08-23:
http://jira.arm.com/browse/IOTSFW-2867

stevew817, 2016-10-11:
At least for NanoStack though, the ping functionality mentioned here is hardcoded in the stack and not exposed upwards. cc @SeppoTakalo

closed 2017-10-27 by ChiefBureaucraticOfficer

nargetdev, 2018-08-20:
Would love to be able to operate directly with MAC addresses for application were reliability is zero tier priority