ARMmbed/mbed-os #2452: Custom targets are broken

jacobrosenthal, 2016-08-15:
Trying to use custom targets as documented says target is not a supported mcu.
```
Jacobs-MacBook-Air-3:BLE_LED jacobrosenthal$ mbed compile -c -t GCC_ARM -m k64f_myapp
usage: make.py [-h] [-m MCU] [-t TOOLCHAIN] [--color] [-c] [-o OPTIONS]
               [-p PROGRAM] [-n PROGRAM] [-j JOBS] [-v] [--silent] [-D MACROS]
               [-S] [-f GENERAL_FILTER_REGEX] [--automated] [--host HOST_TEST]
               [--extra EXTRA] [--peripherals PERIPHERALS]
               [--dep DEPENDENCIES] [--source SOURCE_DIR]
               [--duration DURATION] [--build BUILD_DIR] [-N ARTIFACT_NAME]
               [-d DISK] [-s SERIAL] [-b BAUD] [-L] [--rtos] [--rpc] [--eth]
               [--usb_host] [--usb] [--dsp] [--fat] [--ublox] [--testlib]
               [-l LINKER_SCRIPT]
make.py: error: argument -m/--mcu: k64f_myapp is not a supported MCU. Supported MCUs are:
ARCH_BLE,              ARCH_BLE_BOOT,         ARCH_BLE_OTA, 
ARCH_GPRS,             ARCH_LINK,             ARCH_LINK_BOOT, 
ARCH_LINK_OTA,         ARCH_MAX,              ARCH_PRO, 
ARM_BEETLE_SOC,        ARM_IOTSS_BEID,        ARM_MPS2_M0, 
ARM_MPS2_M0P,          ARM_MPS2_M1,           ARM_MPS2_M3, 
ARM_MPS2_M4,           ARM_MPS2_M7,           B96B_F446VE, 
BLUEPILL_F103C8,       DELTA_DFCM_NNN40,      DELTA_DFCM_NNN40_BOOT, 
DELTA_DFCM_NNN40_OTA,  DISCO_F051R8,          DISCO_F100RB, 
DISCO_F303VC,          DISCO_F334C8,          DISCO_F401VC, 
DISCO_F407VG,          DISCO_F429ZI,          DISCO_F469NI, 
DISCO_F746NG,          DISCO_L053C8,          DISCO_L476VG, 
EFM32GG_STK3700,       EFM32HG_STK3400,       EFM32LG_STK3600, 
EFM32PG_STK3401,       EFM32WG_STK3800,       EFM32ZG_STK3200, 
ELEKTOR_COCORICO,      ELMO_F411RE,           HEXIWEAR, 
HRM1017,               HRM1017_BOOT,          HRM1017_OTA, 
K20D50M,               K22F,                  K64F, 
KL05Z,                 KL25Z,                 KL26Z, 
KL27Z,                 KL43Z,                 KL46Z, 
LPC1114,               LPC11C24,              LPC11U24, 
LPC11U24_301,          LPC11U34_421,          LPC11U35_401, 
LPC11U35_501,          LPC11U35_501_IBDAP,    LPC11U35_Y5_MBUG, 
LPC11U37H_401,         LPC11U37_501,          LPC11U68, 
LPC1347,               LPC1549,               LPC1768, 
LPC2368,               LPC2460,               LPC4088, 
LPC4088_DM,            LPC4330_M0,            LPC4330_M4, 
LPC4337,               LPC810,                LPC812, 
LPC824,                LPCCAPPUCCINO,         MAX32600MBED, 
MAXWSNENV,             MICRONFCBOARD,         MOTE_L152RC, 
MTS_DRAGONFLY_F411RE,  MTS_GAMBIT,            MTS_MDOT_F405RG, 
MTS_MDOT_F411RE,       NRF51822,              NRF51822_BOOT, 
NRF51822_OTA,          NRF51822_Y5_MBUG,      NRF51_DK, 
NRF51_DK_BOOT,         NRF51_DK_LEGACY,       NRF51_DK_OTA, 
NRF51_DONGLE,          NRF51_DONGLE_BOOT,     NRF51_DONGLE_OTA, 
NRF51_MICROBIT,        NRF51_MICROBIT_B,      NRF51_MICROBIT_BOOT, 
NRF51_MICROBIT_B_BOOT, NRF51_MICROBIT_B_OTA,  NRF51_MICROBIT_OTA, 
NRF52_DK,              NUCLEO_F030R8,         NUCLEO_F031K6, 
NUCLEO_F042K6,         NUCLEO_F070RB,         NUCLEO_F072RB, 
NUCLEO_F091RC,         NUCLEO_F103RB,         NUCLEO_F207ZG, 
NUCLEO_F302R8,         NUCLEO_F303K8,         NUCLEO_F303RE, 
NUCLEO_F334R8,         NUCLEO_F401RE,         NUCLEO_F410RB, 
NUCLEO_F411RE,         NUCLEO_F429ZI,         NUCLEO_F446RE, 
NUCLEO_F446ZE,         NUCLEO_F746ZG,         NUCLEO_F767ZI, 
NUCLEO_L011K4,         NUCLEO_L031K6,         NUCLEO_L053R8, 
NUCLEO_L073RZ,         NUCLEO_L152RE,         NUCLEO_L432KC, 
NUCLEO_L476RG,         NUMAKER_PFM_NUC472,    NZ32_SC151, 
OC_MBUINO,             RBLAB_BLENANO,         RBLAB_BLENANO_BOOT, 
RBLAB_BLENANO_OTA,     RBLAB_NRF51822,        RBLAB_NRF51822_BOOT, 
RBLAB_NRF51822_OTA,    RZ_A1H,                SAMD21G18A, 
SAMD21J18A,            SAMG55J19,             SAML21J18A, 
SAMR21G18A,            SEEED_TINY_BLE,        SEEED_TINY_BLE_BOOT, 
SEEED_TINY_BLE_OTA,    SSCI824,               STM32F3XX, 
STM32F407,             TEENSY3_1,             TY51822R3, 
TY51822R3_BOOT,        TY51822R3_OTA,         UBLOX_C027, 
UBLOX_C029,            VK_RZ_A1H,             WALLBOT_BLE, 
WALLBOT_BLE_BOOT,      WALLBOT_BLE_OTA,       WIZWIKI_W7500, 
WIZWIKI_W7500ECO,      WIZWIKI_W7500P,        XADOW_M0, 
XBED_LPC1768           
[mbed] ERROR: "python" returned error code 2.
[mbed] ERROR: Command "python -u /Users/jacobrosenthal/mbed-os-example-ble/BLE_LED/mbed-os/tools/make.py -t GCC_ARM -m k64f_myapp --source . --build ./.build/k64f_myapp/GCC_ARM -c" in "/Users/jacobrosenthal/mbed-os-example-ble/BLE_LED"
---
```

Theres an existing PR at https://github.com/ARMmbed/mbed-os/pull/2196

markus-becker-tridonic-com, 2016-08-18:
@MarceloSalazar If this could get fixed, it would help us a lot.

markus-becker-tridonic-com, 2016-08-18:
FYI @ivan-adzic-tridonic-com

MarceloSalazar, 2016-08-18:
cc @bogdanm @0xc0170 @theotherjimmy

sg-, 2016-09-12:
We're stepping back from custom targets at the moment. Will look at this again in the future.

closed 2016-09-12 by sg-

andrewleech, 2017-04-03:
I'm incredibly disappointed this was removed. I've spent a lot of time getting this going, only to update mbed-os and have it ripped out from under me.

Sure there were issues but now I'm left with no way of customizing the target name without maintaining a fork of mbed-os; which I might as well just downgrade to before this PR :-/

At the very least if these kind of changes are going to be made the docs need be updated to suit:
https://docs.mbed.com/docs/mbedmicro-api/en/latest/config_system/

I'm trying to promote mbed at my workplace but the fragmentation of docs/vs implementation and features being added then removed ad-hoc (like the entire USB stack) are making it very difficult to convince others that it's production worthy...

0xc0170, 2017-04-03:
> Sure there were issues but now I'm left with no way of customizing the target name without maintaining a fork of mbed-os; which I might as well just downgrade to before this PR :-/

cc @theotherjimmy @screamerbg

> At the very least if these kind of changes are going to be made the docs need be updated to suit:
https://docs.mbed.com/docs/mbedmicro-api/en/latest/config_system/

@AnotherButler

> I'm trying to promote mbed at my workplace but the fragmentation of docs/vs implementation and features being added then removed ad-hoc (like the entire USB stack) are making it very difficult to convince others that it's production worthy...

I would like to find out more about the issues you are facing, some of those we are aware of and addressing. Please create a new issue to summarize all these points.

AnotherButler, 2017-04-03:
@0xc0170 I shall put in a request to have that page redirect to the updated document, https://docs.mbed.com/docs/mbed-os-handbook/en/latest/advanced/config_system/. Could you please review the updated document at this link and let me know what content requires updating?

andrewleech, 2017-04-03:
@AnotherButler that link looks correct! 

Sorry I posted late in the day when I was a bit frustrated because I've been looking for methods of defining a new target based on an existing one that doesn't involve forking mbed-os but following the docs on custom targets was exactly what I wanted. But when it was first throwing weird errors, then I found out it had been removed it just got the better of me.

I and a couple of others on a new team are having difficulties finding correct docs for lots of the mbed-os and tooling, mostly because google searches show such a mix of docs from mbed 2, 3 and 5. Or the same information on multiple different pages, some more up to date than others.

The usb issue I mentioned was based on ordering some (advertised) mbed supported dev boards (32F429IDISCOVERY) with onboard usb, only to find the usb stack has been made unsupported since I last used mbed a couple of years ago, and said unsupported stack doesn't work on this mbed board. I had others in the team wanting to switch to the ugly auto-generated flat c code from st's cube tool in preference to mbed due to issues like this, they got the impression mbed is too new and unproven to trust in a production environment. 
Please don't take this the wrong way though I know this is community supported and I absolutely love the infrastructure you've built here, the mbed cli tool is amazing! I'll go harass ST to update their USBDevice lib :-) 

I'm not sure I have any other specific concerns in mind, I'll raise new issues if/when I run into problems thanks.

theotherjimmy, 2017-04-03:
>  I'll raise new issues if/when I run into problems 

Please do. We'll be sure to help you out in the future!

andrewleech, 2017-04-05:
@AnotherButler Thanks, I also since realized lots of our confusion was caused by not noticing that the docs for all versions are available online (with version in the url).

In my opinion it'd be useful if all online docs version older than latest made mention of this the similar to the mbed 2 pages.
You can't stop google linking to older version of docs for sure, but at making it obvious to the user that they might not be on the current ones is helpful.