ARMmbed/mbed-os #2459: Cannot enable Debug on generated Makefile

MarceloSalazar, 2016-08-16:
Tools version:
mbed CLI = 0.9.1
project_generator = 0.9.9

I expect to be able to enable DEBUG on the generated Makefile by defining `DEBUG = 1` before this `ifeq`:

```
ifeq ($(DEBUG), 1)
  CC_FLAGS += -DDEBUG -O0
else
  CC_FLAGS += -DNDEBUG -Os
endif
```

However, debug symbols are not enabled because:
1. There is no option to add `-DDEBUG -O0` to `CPPC_FLAGS` in the `ifeq` statement (e.g. `main.cpp`)
2. `CC_FLAGS` and `CPPC_FLAGS` are already defined earlier with `-Os` option:
```
CC_FLAGS = -c ... -Os ...
CPPC_FLAGS = -c ... -Os ...
```

Note: ticket originally raised here: https://github.com/project-generator/project_generator/issues/438

@0xc0170 @theotherjimmy

theotherjimmy, 2016-08-16:
From the previous discussion:
>Actually, the flag to add debug information to GCC's output files is `-g` and it is simply missing from the CC_FLAGS variable. Could you try adding `-g` to `CC_FLAGS` in the "true" part of that if statement, and tell me if you can see the debug symbols?

theotherjimmy, 2016-08-19:
If you replace the block you mentioned above with:
```
ifeq ($(DEBUG), 1)
  CC_FLAGS += -DDEBUG -O0 -g
  CPPC_FLAGS += -DDEBUG -O0 -g
else
  CC_FLAGS += -DNDEBUG -Os
  CPPC_FLAGS += -DNDEBUG -Os
endif
```
and call make with `make DEBUG=1`
You should be able to debug both c and c++

sg-, 2016-08-25:
@MarceloSalazar Any feedback about the requested ```-g``` addition?

MarceloSalazar, 2016-08-26:
Sorry for late response.
I confirm the changes suggested by @theotherjimmy work and I'm able to debug the main (cpp) app.
It would be great if these could be added to the default generated Makefile.

theotherjimmy, 2016-08-26:
Awesome. That's a quick change. I'll get a PR in soon.

theotherjimmy, 2016-09-15:
@MarceloSalazar, the new makefiles are a bit cleaner, and have debug-info enabled by default.

MarceloSalazar, 2016-09-16:
Thanks, it looks good :)

MarceloSalazar, 2016-09-20:
Yes, I tested the master version and works ok. Closing.

closed 2016-09-20 by MarceloSalazar