ARMmbed/mbed-os #2460: Makefile - lpc1768 does not build

0xc0170, 2016-08-16:
``python tools\project.py -m LPC1768 -i gcc_arm -n MBED_10 -b``

Error output:
```
arm-none-eabi-g++ -c -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -fno-builtin -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks -fomit-frame-p
ointer -mcpu=cortex-m3 -mthumb -Os -std=gnu++98 -fno-rtti -Wvla -DDEVICE_ERROR_PATTERN=1 -D__MBED__=1 -DDEVICE_I2CSLAVE=
1 -DTARGET_LIKE_MBED -DTARGET_NXP -DTARGET_LPC176X -D__MBED_CMSIS_RTOS_CM -DDEVICE_RTC=1 -DTOOLCHAIN_object -DMBED_USERN
AME=Kojto -D__CMSIS_RTOS -DTOOLCHAIN_GCC -DDEVICE_CAN=1 -DTARGET_LIKE_CORTEX_M3 -DTARGET_CORTEX_M -DARM_MATH_CM3 -DDEVIC
E_ANALOGOUT=1 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_M3 -DDEVICE_PWMOUT=1 -DDEVICE_INTERRUPTIN=1 -DDEVICE_I2C=1 -DDEVICE_P
ORTOUT=1 -D__CORTEX_M3 -DDEVICE_STDIO_MESSAGES=1 -DTARGET_LPC1768 -DTARGET_RELEASE -DDEVICE_PORTINOUT=1 -DDEVICE_SERIAL_
FC=1 -DMBED_BUILD_TIMESTAMP=1471343151.62 -DTARGET_MBED_LPC1768 -DDEVICE_PORTIN=1 -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -
DDEVICE_SPI=1 -DDEVICE_ETHERNET=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_SERIAL=1 -DDEVICE_SEMIHOST=1 -DDEVICE
_DEBUG_AWARENESS=1 -DDEVICE_LOCALFILESYSTEM=1 -include mbed_config.h -MMD -MP  -I. -I./env -I./mbed -I./mbed/TARGET_LPC1
768 -I./mbed/TARGET_LPC1768/TARGET_NXP -I./mbed/TARGET_LPC1768/TARGET_NXP/TARGET_LPC176X -I./mbed/TARGET_LPC1768/TARGET_
NXP/TARGET_LPC176X/TARGET_MBED_LPC1768 -I./mbed/TARGET_LPC1768/TOOLCHAIN_GCC_ARM -I.  -o main.o ../main.cpp
process_begin: CreateProcess(NULL, arm-none-eabi-g++ -c -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializ
ers -fmessage-length=0 -fno-exceptions -fno-builtin -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete
-null-pointer-checks -fomit-frame-pointer -mcpu=cortex-m3 -mthumb -Os -std=gnu++98 -fno-rtti -Wvla -DDEVICE_ERROR_PATTER
N=1 -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DTARGET_NXP -DTARGET_LPC176X -D__MBED_CMSIS_RTOS_CM -DDEVICE_RT
C=1 -DTOOLCHAIN_object -DMBED_USERNAME=Kojto -D__CMSIS_RTOS -DTOOLCHAIN_GCC -DDEVICE_CAN=1 -DTARGET_LIKE_CORTEX_M3 -DTAR
GET_CORTEX_M -DARM_MATH_CM3 -DDEVICE_ANALOGOUT=1 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_M3 -DDEVICE_PWMOUT=1 -DDEVICE_INTE
RRUPTIN=1 -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -D__CORTEX_M3 -DDEVICE_STDIO_MESSAGES=1 -DTARGET_LPC1768 -DTARGET_RELEASE -D
DEVICE_PORTINOUT=1 -DDEVICE_SERIAL_FC=1 -DMBED_BUILD_TIMESTAMP=1471343151.62 -DTARGET_MBED_LPC1768 -DDEVICE_PORTIN=1 -DD
EVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DDEVICE_SPI=1 -DDEVICE_ETHERNET=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_SE
RIAL=1 -DDEVICE_SEMIHOST=1 -DDEVICE_DEBUG_AWARENESS=1 -DDEVICE_LOCALFILESYSTEM=1 -include mbed_config.h -MMD -MP -I. -I.
/env -I./mbed -I./mbed/TARGET_LPC1768 -I./mbed/TARGET_LPC1768/TARGET_NXP -I./mbed/TARGET_LPC1768/TARGET_NXP/TARGET_LPC17
6X -I./mbed/TARGET_LPC1768/TARGET_NXP/TARGET_LPC176X/TARGET_MBED_LPC1768 -I./mbed/TARGET_LPC1768/TOOLCHAIN_GCC_ARM -I. -
o main.o ../main.cpp, ...) failed.
make (e=2): The system cannot find the file specified.
make[1]: *** [main.o] Error 2
make: *** [all] Error 2
```

Looks like the paths are wrong (VPATH is set to ..) and -I paths are referencing not valid paths.

cc @sarahmarshy @theotherjimmy

sarahmarshy, 2016-08-16:
I wasn't able to replicate this issue.

theotherjimmy, 2016-08-16:
I was not able to reproduce this issue.

0xc0170, 2016-08-17:
False alarm, my paths were corrupted thus arm-none-eabi-gcc was not found  - I found once I was trying to use gcc in the command line (in the above was g++). Can we add a check if gcc/g++ is in present and fail prematurely in the Makefile?

theotherjimmy, 2016-08-18:
> Can we add a check if gcc/g++ is in present and fail prematurely in the Makefile?

From a performance perspective, this is not going to make any difference. This is actually a bit difficult, as there is no cross platform which/where and the errors returned on multiple platforms are very different.

sarahmarshy, 2016-08-18:
@0xc0170 we can do this within mbed:

https://github.com/ARMmbed/mbed-os/pull/2418

theotherjimmy, 2016-08-18:
This is after an export, so that PR is irrelevant.

closed 2016-08-18 by 0xc0170