ARMmbed/mbed-os #2470: freopen with NULL filepath fails

aamcampbell, 2016-08-16:
According to the C standard for freopen() (as stated in POSIX),

"If filename is a null pointer, the freopen() function shall attempt to change the mode of the stream to that specified by mode, as if the name of the file currently associated with the stream had been used."
(http://pubs.opengroup.org/onlinepubs/009695399/functions/freopen.html)

However, when I call freopen(NULL, "mode", pFile) in ARM mbed, it fails and returns NULL. The call fails whether or not the mode is the same as pFile was originally opened with.

0xc0170, 2016-08-17:
Can you specify the toolchain + target you use?

aamcampbell, 2016-08-17:
Sorry, I meant to include that in my original report:
yotta (mbed OS 3) with arm GCC on the FRDM-K64F.

sg-, 2016-08-25:
Can you provide a test case in the form of a program we can work against?

aamcampbell, 2016-08-26:
Sure! Reliance Edge is the only filesystem that I know officially supports mbed OS 3, and it's proprietary; so I figured out how to do it with mbed 5 CLI instead. The issue appears there as well.

Here

- Install mbed CLI.
- Run these commands to download an SD card example program. (Update mbed to avoid #1754.)

```
mbed import http://mbed.org/users/neilt6/code/SDFileSystem_HelloWorld/
cd SDFileSystem_HelloWorld/mbed
mbed update
cd ../SDFileSystem/FATFileSystem
mbed update
cd ../../
```

- Open SDFileSystem_HelloWorld/main.cpp.
- Change lines 5-6 to the following:

```
DigitalIn button(PTC6, PullUp);
SDFileSystem sd(PTE3, PTE1, PTE2, PTE4, "sd");
```

- At the top of the file, add:

```
#include <cassert>
#include <cstdio>
```

- After line ~96 (`printf("success!\n");`), add:

```
        FILE *f=fopen("/sd/file", "w");
        assert(f != NULL);
        f = freopen(NULL, "w", f);
        assert(f != NULL);
```

- Compile: `mbed compile -t GCC_ARM -m K64F --options debug-info`
- Insert FAT-formatted SD card to the K64F, start debugging, and press SW2 after a second. If FAT mounts fine, it will crash on the assert() after freopen().

closed 2017-10-27 by ChiefBureaucraticOfficer