ARMmbed/mbed-os #2474: [TARGET_NRF5] serial_api.c looks not async even if there are DEVICE_SERIAL_ASYNCH=1 macro

ytsuboi, 2016-08-17:
I will try to check the code later, but this serial code requires CTS input. When printf() stopped,  we can solve that by touching CTS pin. @anangl

ytsuboi, 2016-08-17:
It hangs at https://github.com/ARMmbed/mbed-os/blob/master/hal/targets/hal/TARGET_NORDIC/TARGET_NRF5/serial_api.c#L290

ytsuboi, 2016-08-17:
Patched. Now, printf() works fine on TY51822r3. Still need to test on nRF51-DK.
https://github.com/ytsuboi/mbed-os/commit/992abd7410248194e9e61def1947e12ce5541b83

Can I make a pull request after testing with nRF51-DK? @anangl @0xc0170

ytsuboi, 2016-08-17:
Oops, I will change a macro to switch.

ytsuboi, 2016-08-18:
PR made. https://github.com/ARMmbed/mbed-os/pull/2482

nuket, 2016-08-18:
This hang is also biting me hard on the nRF51-DK.

ytsuboi, 2016-08-19:
@nuket Could you provide the detail? How this issue biting you?

ytsuboi, 2016-08-19:
https://github.com/ARMmbed/mbed-os/issues/2489

sg-, 2016-08-25:
Can you confirm is #2535 resolved this issue?

ytsuboi, 2016-08-26:
PR made https://github.com/ARMmbed/mbed-os/pull/2555

nvlsianpu, 2016-08-30:
@ytsuboi Could you close this issue?

ytsuboi, 2016-08-30:
Solved! @nvlsianpu

closed 2016-08-30 by ytsuboi