ARMmbed/mbed-os #2475: Mbed5 memory profiling

javier-moreno-tridonic-com, 2016-08-17:
Note: This is just a template, so feel free to use/remove the unnecessary things

## Description

Type: Question
Priority: Minor
Question
---
On mbed 3 the "ualloc" yotta module was used for *malloc* and *free* implementation. This allowed us to enable and customise an UART trace, allowing us to study the dynamic memory performance our code and look for optimisations. E.g.

```C++
void *mbed_ualloc(size_t bytes, UAllocTraits_t traits, void *cb_caller)
{
	void *caller;
	if (cb_caller == NULL) {
		caller = (void*)caller_addr();
	} else {
		caller = cb_caller;
	}
    void *p = mbed_ualloc_internal(bytes, traits, caller);
    if (!prevent_tracing) {
        prevent_tracing = 1;
        ualloc_debug(UALLOC_DEBUG_LOG, "ua c:%p s:%u m:%p\n", caller, (unsigned)bytes, p);
        prevent_tracing = 0;
    }
    return p;
}
```

So far I could not figure out how to replicate this in mbed5. Is this memory tracing still a possibility?

@MarceloSalazar @markus-becker-tridonic-com

MarceloSalazar, 2016-08-17:
This is related to the conversation happening on this PR: https://github.com/ARMmbed/mbed-os/pull/2442

sg-, 2016-08-25:
#2442 has been merged. This can be enabled by using ```-D MBED_HEAP_STATS_ENABLED``` macro at compile time

javier-moreno-tridonic-com, 2016-08-26:
Well my question/requests was more in the line of what is being disscussed in https://github.com/ARMmbed/mbed-os/pull/2487

In there a time frame for this PR?

sg-, 2016-08-26:
Any day now. @bogdanm is out for a few days but I think @0xc0170 will rebase and get this ready for merge.

closed 2016-08-26 by sg-