ARMmbed/mbed-os #2483: CFStore: SRAM limitation

markus-becker-tridonic-com, 2016-08-18:
According to the document https://github.com/ARMmbed/mbed-os/blob/1e069aa00963a158341ea0554499d65275e73ec9/features/storage/FEATURE_STORAGE/cfstore/doc/design/configuration_store_app_note_0001.md, chapter `### SRAM Footprint, the current CFSTORE implementation` has the limitation that the maximum size of data stored in flash is limited by the maximum amount of SRAM that can be used to store the key value pairs prior to being flushed to flash and this limitation will be removed in a future implementation.

On the other hand according to the document https://github.com/ARMmbed/mbed-os/blob/1e069aa00963a158341ea0554499d65275e73ec9/features/storage/FEATURE_STORAGE/cfstore/doc/design/configuration_store_hld.md, chapter `## CFSTORE Client` Writes KV Value Blob Size > Available SRAM, a write of a key value pair greater than the available SRAM seems to be possible by splitting the write into several smaller fragments less than the available SRAM.

This seems a little contradictory. Thus the question is if there really is a limitation regarding the amount of storable configuration data with the current CFSTORE implementation or not? If so, when is the limitation to be resolved?

@MarceloSalazar

MarceloSalazar, 2016-08-19:
@simonqhughes please review this

sg-, 2016-08-25:
The high level design aims to chunk kv pairs where the required RAM << NV STORAGE. The current implementation does not meet the aspirations of the high level design.

I'd suggest this issue gets filed against https://github.com/ARMmbed/configuration-store/issues

simonqhughes, 2016-08-30:
Hi,

Yes, the 
```
CFSTORE Client  Writes KV Value Blob Size > Available SRAM,
```
section is misleading and should be re-written, which I will do in due course. Please ignore the information in this section at this time.

Thanks
Simon

daniel-cesarini-tridonic-com, 2016-08-30:
Hi there,

Is there a plan to remove this limitation in future? 
If yes, by when should this happen?

Thank you,
Daniel.

simonqhughes, 2016-09-01:
Hi
Yes, the plan is to remove this limitation in the future. As to when, this will depend on priorities. The priority at present are:
- provide CFSTORE C++ interface and provide good integration with mbedOS.
- provide CFSTORE-uvisor integration to secure the stored data .

Best
Simon

MarceloSalazar, 2016-09-06:
Marcelo to discuss internall

sg-, 2016-09-06:
please open this on CFStore

MarceloSalazar, 2016-09-20:
Leaving this ticket open to keep track of the request, as cannot access to this ticket:
https://github.com/ARMmbed/configuration-store/issues/26

closed 2017-10-27 by ChiefBureaucraticOfficer