ARMmbed/mbed-os #2486: Build config system broken

tommikas, 2016-08-18:
Seems that a [recent commit](https://github.com/ARMmbed/mbed-os/commit/869b5d7813909f9bcdcd8f3837284e4a0c2a7146) broke the build configuration system.

### Before the merge:

The macro section in the generated mbed_config.h:
```
// Macros
#define NS_USE_EXTERNAL_MBED_TLS                                                                                                                                  // defined by library:nanostack
#define MEM_ALLOC                                                malloc                                                                                           // defined by application
#define UNITY_INCLUDE_CONFIG_H                                                                                                                                    // defined by library:utest
#define MEM_FREE                                                 free        
```

The build result output:
```
Link: mbed-os-cliapp
Elf2Bin: mbed-os-cliapp
+-----------------------------+--------+-------+-------+
| Module                      |  .text | .data |  .bss |
+-----------------------------+--------+-------+-------+
| Fill                        |    516 |    10 |  2507 |
| Misc                        | 113495 |  2341 |  3918 |
| features/FEATURE_CLIENT     |  66692 |     3 |    53 |
| features/FEATURE_COMMON_PAL |  19044 |    89 |  8404 |
| features/frameworks         |   3183 |    52 |   328 |
| features/mbedtls            | 117016 |    51 |  8867 |
| features/net                |  35626 |   102 | 51259 |
| hal/common                  |   3197 |    20 |   305 |
| hal/targets                 |  15272 |    12 |   200 |
| rtos/rtos                   |    773 |     4 |     0 |
| rtos/rtx                    |   7421 |    20 |  6787 |
| Subtotals                   | 382235 |  2704 | 82628 |
+-----------------------------+--------+-------+-------+
Allocated Heap: 65540 bytes
Allocated Stack: 32768 bytes
Total Static RAM memory (data + bss): 85332 bytes
Total RAM memory (data + bss + heap + stack): 183640 bytes
Total Flash memory (text + data + misc): 385979 bytes
Image: ./.build/K64F/GCC_ARM/mbed-os-cliapp.bin
```

### After the merge:

The macro section in the generated mbed_config.h:
```
// Macros
#define MEM_FREE                                                                                                                                                  // defined by application
#define MEM_FREE                                                 free                                                                                             // defined by application
#define MEM_FREE                                                                                                                                                  // defined by application
#define MEM_FREE                                                 free      
```

Warnings generated for every file:
```
[Warning] mbed_config.h@50,1: "MEM_FREE" redefined
[Warning] mbed_config.h@51,1: "MEM_FREE" redefined
[Warning] mbed_config.h@52,1: "MEM_FREE" redefined
```

The build result output:
```
Link: mbed-os-cliapp
Elf2Bin: mbed-os-cliapp
+-----------+--------+-------+-------+
| Module    |  .text | .data |  .bss |
+-----------+--------+-------+-------+
| Subtotals | 382267 |  2704 | 82628 |
+-----------+--------+-------+-------+
Allocated Heap: 65540 bytes
Allocated Stack: 32768 bytes
Total Static RAM memory (data + bss): 85332 bytes
Total RAM memory (data + bss + heap + stack): 183640 bytes
Total Flash memory (text + data + misc): 386011 bytes
Image: ./.build/K64F/GCC_ARM/mbed-os-cliapp.bin
```

And of course the application doesn't function properly.

bogdanm, 2016-08-18:
@theotherjimmy

jupe, 2016-08-18:
so it related to this PR: https://github.com/ARMmbed/mbed-os/pull/2458

theotherjimmy, 2016-08-18:
I was able to reproduce this issue. I'm debugging it now.

theotherjimmy, 2016-08-18:
I was able to fix the problem. PR coming.

theotherjimmy, 2016-08-18:
I think there is another issue related to the printing of only the subtotals. I'll look into that too.

jupe, 2016-08-18:
good found 👍

theotherjimmy, 2016-08-18:
memap printing issue: https://github.com/ARMmbed/mbed-os/pull/2494

closed 2016-08-19 by 0xc0170

jupe, 2016-08-19:
@tommikas could you verify that this really works in master now?

PR #2492 (2016-08-19): Fix macros being the exact same when generating mbed_config.h
resolves #2486