ARMmbed/mbed-os #2500: Allow for tests in application repositories

markus-becker-tridonic-com, 2016-08-19:
In the [mbed-cli documentation](https://github.com/ARMmbed/mbed-cli#compiling-and-running-tests) it is written:

> Note: This feature does not work in applications that contain a main function that is outside of a TESTS directory.

Would it be possible to resolve this limitation? This way we would not need to have a separate repository for tests with code that is duplicated in the actual implementation and in the tests. (Would rather not use .lib files to pull it in.

@MarceloSalazar @daniel-cesarini-tridonic-com @ivan-adzic-tridonic-com

markus-becker-tridonic-com, 2016-08-19:
When compiling tests, the code in the main directory of the repository could be ignored, or?

0xc0170, 2016-08-19:
cc @bridadan @mazimkhan

MarceloSalazar, 2016-08-19:
This is a valid use case and should be considered as enhancement in our tools.
@screamerbg @sg-

sg-, 2016-09-19:
Look for conversation about test injected macro and alternate main entry point

MarceloSalazar, 2016-09-20:
Could you please link to that conversation?

sg-, 2016-09-20:
Its on the developer site. Not having much luck with the search there right now

markus-becker-tridonic-com, 2016-10-24:
The workaround sort of works, but still it would be better to have this solved in mbed test properly. The workaround requires adding #ifdef's in every source file.

closed 2017-10-27 by ChiefBureaucraticOfficer