ARMmbed/mbed-os #2506: Exporters - SWSTEM32 fpu problem

0xc0170, 2016-08-22:
Reported here : https://developer.mbed.org/questions/73309/mbed-os-basic-example-compile-error/#answer10867

Seems that fpu is not correctly set for exported project, a resolution provided by Graham Sayers.

@bcostm @adustm @LMESTM

sg-, 2016-09-19:
@bcostm @adustm @LMESTM bump

bcostm, 2016-09-23:
I have exported the Nucleo_RTOS_basic example on SW4STM32 toolchain using Nucleo_F446RE platform. I confirm the build error.

I have also change the FPU setting in my project to: 
MCU Settings Flaoting point hardware: fpv4-sp-d16
Floating-point ABI: softfp
Instruction Set: Thumb II
As proposed Graham, and the build and execution are OK

Now, I looked in mbed-os/tools/export folder and there is only one SW4STM32 cproject template common to all platforms: `sw4stm32_cproject_common.tmpl`. In this template there is nothing concerning the FPU options, this is why the FPU options are empty when the project is generated.

So, this template file must be changed in order to configure the FPU settings differently for each platform. The sw4stm32.py file will have to be changed also...

bcostm, 2016-09-23:
I have already something working. I just need now to check all platforms and set the correct FPU option in the sw4stm32.py file. The PR will come soon...

closed 2017-01-17 by sg-