ARMmbed/mbed-os #2507: Export to GCC @ Online Compiler

andcor02, 2016-08-22:
Received a few flags from users that the exporting of programs from the web and online compiler are not possible when choosing to export to GCC(ARM Embedded). I have also replicated the problem.

In summary the issue seems to be:

- Cannot export to GCC using web or online compiler (All other IDEs operational)
- Seems to only effect mbed-os and its libraries (working with mbed 2.0)
- Irrespective of platform

Regards,

Andrea, team mbed

sg-, 2016-08-23:
Thanks for the report. Exporters are being refactored right now and we hope to have these issues corrected soon.

Testato, 2016-08-29:
I confirm this issue 
(Win10 - Nucleo F 103RB)

theotherjimmy, 2016-08-29:
@Testato Thanks for reproducing the issue on your end. We're working on it!

theotherjimmy, 2016-09-09:
After #2245 merged, we are working on updating the exporters on the website should be upgraded with the next release.

sg-, 2016-09-19:
@andcor02 Can this be closed?

andcor02, 2016-09-19:
sg- yes this can be closed

closed 2016-09-19 by andcor02